mne-tools/mne-python #4122: Consistency in plot_topomap APIs

hubertjb, 2017-03-29:
The `evoked.plot_topomap` function has parameters `times` and `average` to specify the times and range to average over:
https://github.com/mne-tools/mne-python/blob/master/mne/evoked.py#L333

At the same time, the `plot_topomap` method in `AverageTFR` has parameters `tmin` and `tmax`:
https://github.com/mne-tools/mne-python/blob/master/mne/time_frequency/tfr.py#L1279

Talking with @jona-sassenhagen, I guess these two APIs should be uniformised?

jona-sassenhagen, 2017-03-29:
See https://github.com/mne-tools/mne-python/issues/4116

:D

agramfort, 2017-03-29:
not clear to me

larsoner, 2018-04-04:
Closing since it's not clear what the unification would be, but feel free to reopen if you have an idea

jona-sassenhagen, 2018-04-04:
Well, `TFR.plot_topomap` could lose `tmin` and `tmax` and gain `times` and `average` instead?

larsoner, 2018-04-04:
Ahh okay, based on what @agramfort said I thought it would be more complicated