mne-tools/mne-python #4124: VIZ: sequential colors for plot_compare_evokeds

jona-sassenhagen, 2017-03-29:
While I'm waiting for the CIs ...

For plotting non-factorial/continuous manipulations (e.g. word frequency, stimulus intensity ...), it would be cool to be able to do something like this:

```
mne.viz.plot_compare_evokeds(evokeds, colors="sequential")
```
[image]

Anybody interested in having this?

agramfort, 2017-03-30:
how do you get this ? with a cmap ? a palette from seaborn?

mmagnuski, 2017-03-30:
@argamfort - that's viridis colormap, you can do it like this:
```python
x = np.linspace(0, 1, num=15)
colors = plt.cm.viridis(x)
```

agramfort, 2017-03-30:
that's what I thought so I would allow to pass a cmap for colors

jona-sassenhagen, 2017-03-30:
Fine with me, so that would mean allowing users to pass a str for `colors` (it's dict/list right now) which would trigger sequential mapping of cond names, e.g.
`plot_compare_evokeds(evokeds, colors='inferno_r')`

And then this would always try to find numbers in the dict keys (the cond names) and order the evokeds based on that.
Or we could just always try to order them based on numeric parts of the names, I guess the code overhead is small.

agramfort, 2017-03-30:
no objection

jona-sassenhagen, 2017-03-30:
@mmagnuski any further suggestions?
Do you think code like this would be useful for people besides me (and @dengemann)?

dengemann, 2017-03-30:
Or directly the cmap object. I'd also draw a color bar as discussed
yesterday.

mmagnuski, 2017-03-30:
I'm ok with passing a colormap or cmap name. But I think it should work for list of erps too (not only for dictionary with keys that have some numbers), WDYT?

Eric89GXL, 2017-03-30:
`list` is already ordered so fine by me

jona-sassenhagen, 2017-03-30:
Ok, so then the only question is how the ordering should be made. The problem is Python doesn't have natural search in main. I'm thinking of something like this:

- attempt to split the cond name by "/"
- for each item in the resulting list, filter for numeric parts
- once one item has len > 0, convert to float and use that as the sort key

So roughly:

`evokeds.keys()` would be `'word/frequency_10', 'word/frequency_30', 'word/frequency_90', 'word/frequency_60'`

To make matters more complicated, you could imagine something like

```
'word/length_4', 'word/length_6', 'word/length_8', 'nonword/length_4', 'nonword/length_6', 'nonword/length_8',
```

Where I'm not perfectly sure what to do, or (worse)

```
'word/length_4', 'word/length_6', 'word/length_8', 'nonword/length_4', 'nonword/length_6', 'nonword/length_8',
```

or (even worse still)

```
'word/length_4'/freq_10', 'word/length_6/freq_9', 'word/length_8/freq_3', 'nonword/length_4/freq_10', 'nonword/length_6/freq_7', 'nonword/length_8/freq_4',
```

@dengemann and the colorbar would be triggered by >9 or so items to be plotted I suggest?
Though this might imply having to map proportionally (e.g. if your bins are not regularly spaced).

Super complication: a factorial plus a long continuous factor, e.g. 

```
linestyles={"word":"--", "nonword":"-"}
colors="viridis"
```

and the cond names:

```
'word/freq_1', 'word/freq_2', ... 'word/freq_100', 'nonword/freq_1', 'nonword/freq_2', ... 'nonword/freq_100''
```

So you'd need a colorbar legend and a legend for linestyle ... (And this is the part where Alex tells me to just style it with matplotlib post hoc.)

jona-sassenhagen, 2017-03-30:
Unless anybody else has a good idea right now, I'll just play around with it a bit and propose a solution.

agramfort, 2017-03-30:
ok but please please KISS !

mmagnuski, 2017-03-30:
@jona-sassenhagen 
I'd restrict the usage for example to using only the last '/'-separated condition that has numbers in it.
If user wants something more complex they would have to create the colors themselves.

jona-sassenhagen, 2017-03-30:
So Alex' suggestion is to just force users to format their keys so that the sort we're doing right now works automatically. E.g., we already sort the keys; users have to zero-pad their keys because we can't do "natural sort". So here'd be a minimal API idea: `colors` param can take a str*. If that is the case, the corresponding colormap is used and the events are mapped to it in a linear manner. (E.g., we don't treat "freq_1, "freq_2, freq_3" any different from "freq_1, freq_2, freq_10".)

* Should also have a colormap option, but maybe for later ...

jona-sassenhagen, 2017-04-12:
What do you think?

Not so good:
```
style_plot = dict(
    colors='viridis',
    linestyles={"Concrete":"-", "Abstract":"-."},
    split_legend=False,
    sequential=True
)
print(list(evokeds.keys()))
mne.viz.evoked.plot_compare_evokeds(evokeds, **style_plot, picks=5);

out: ['NumberOfLetters_04/Concrete', 'NumberOfLetters_06/Abstract', 'NumberOfLetters_07/Abstract', 'NumberOfLetters_06/Concrete', 'NumberOfLetters_07/Concrete', 'NumberOfLetters_08/Concrete', 'NumberOfLetters_04/Abstract', 'NumberOfLetters_05/Concrete', 'NumberOfLetters_05/Abstract', 'NumberOfLetters_08/Abstract']
```
[image]

Better:

```
style_plot = dict(
    colors='viridis',
    linestyles={"Concrete":"-", "Abstract":"-."},
    split_legend=True,
    sequential=True
)
print(list(evokeds.keys()))
mne.viz.evoked.plot_compare_evokeds(evokeds, **style_plot, picks=5);

out: ['NumberOfLetters_04/Concrete', 'NumberOfLetters_06/Abstract', 'NumberOfLetters_07/Abstract', 'NumberOfLetters_06/Concrete', 'NumberOfLetters_07/Concrete', 'NumberOfLetters_08/Concrete', 'NumberOfLetters_04/Abstract', 'NumberOfLetters_05/Concrete', 'NumberOfLetters_05/Abstract', 'NumberOfLetters_08/Abstract']
```

[image]

And that also allows:

```
evokeds = mne.read_evokeds(sample.data_path() + '/MEG/sample/sample_audvis-ave.fif',
                          baseline=(None, 0))
evokeds = {"/".join(evoked.comment.split()):evoked.pick_types(meg='grad') for evoked in evokeds}

style_plot = dict(
    colors={"Auditory":"crimson", "visual":"cornflowerblue"},
    linestyles={"Left":"-", "Right":":"},
    split_legend=True
)
print(list(evokeds.keys()))
mne.viz.evoked.plot_compare_evokeds(evokeds, **style_plot);

out: ['Right/Auditory', 'Left/visual', 'Right/visual', 'Left/Auditory']
```

[image]

@mmagnuski @agramfort @dengemann

dengemann, 2017-04-12:
The colorbar rocks!

jona-sassenhagen, 2017-04-12:
Sadly it essentially makes Alex' minimalist suggestion impossible - we need to extract numbers from the dict key string names, remove duplicates etc..

jona-sassenhagen, 2017-04-12:
Btw the dataset I'm using is freely available - [it is on OSF](https://osf.io/huzs4/) - and a good case for "rich" epoch annotation, possibly to be added to the sample data (epochs fif is < 30 MB):

WORD#|WORD|Concreteness|WordFrequency|OrthographicDistance|NumberOfLetters|BigramFrequency|ConsonantVowelProportion|VisualComplexity
---|---|---|---|---|---|---|---|---
WORD1|film|5.45|3.18949|1.75|4.0|343.25|0.75|55.78371
WORD2|cent|5.9|3.700704|1.35|4.0|546.75|0.75|63.141553
WORD3|shot|4.6|2.858537|1.2|4.0|484.75|0.75|64.600033
WORD4|cold|3.7|3.45454|1.15|4.0|1095.25|0.75|63.657456999999994
WORD5|main|3.0|3.539076|1.35|4.0|686.0|0.5|68.945661

agramfort, 2017-04-13:
@jona-sassenhagen please share a gist to process these data so we can have a look

jona-sassenhagen, 2017-04-13:
http://nbviewer.jupyter.org/gist/jona-sassenhagen/b1fca57f72e36b8c225d0315b18a0206 @agramfort

jona-sassenhagen, 2017-04-18:
@agramfort have you had time to look at it?

agramfort, 2017-04-18:
I looked but had no time to play :(

jona-sassenhagen, 2017-04-20:
Btw @mmagnuski , have you looked at the Gist? I can imagine you have some opinions here.

mmagnuski, 2017-04-20:
@jona-sassenhagen Yes, I have - it looks really convenient. The only other opinion I have is that I prefer dashed to dotted lines. :)

closed 2018-03-05 by jona-sassenhagen