mne-tools/mne-python #4125: Extending the Annotations object

choldgraf, 2017-03-29:
It seems a few of us have had conversations about how to include more rich metadata about the world in MNE objects. Currently we only have events objects, which give us only onset + event ID, as well as `Annotations` which tell us onset + duration + a string. There are many instances where we want more than just a string to describe one event.

Why not allow for the `Annotations` class to take a list of dictionaries as well, where each dictionary could have an arbitrary number of `key: val` pairs that would describe metadata for a given event. You could ensure that keys are always strings, and the values are one of `str, int, float, date` or something like this.

Then you could write a `to_dataframe` function that would build a pandas dataframe using the information that's in `Annotations`, which would let you do queries in intelligent ways. In a future PR, you could also imagine writing some kind of `query` method that would satisfy the query conditions and return a list of which Annotations match this, and could be easily exportable to an MNE events object.

WDYT? @kingjr @agramfort ?

Eric89GXL, 2017-03-30:
`Annotations` are read/written from/to FIF. So Dataframe conversion/use is probably a no-go.

choldgraf, 2017-03-30:
so that means that there's just no possibility for a more rich representation of metadata for task data, other than having lots and lots of event codes?

choldgraf, 2017-03-30:
@kingjr can you send me a link to the MEG data you were talking about? I can try to set up a use-case example

kingjr, 2017-03-30:
You mean MNE-HCP? You have to create an account on HCP and aws :/ maybe
this isn't appropriate actually. cc @dengemann

Eric89GXL, 2017-04-18:
@jona-sassenhagen didn't you recently post a link to some data, or description of some data, that would work well for the "rich event description"? I tried searching my emails but failed to find it.

agramfort, 2017-04-18:
here

http://nbviewer.jupyter.org/gist/jona-sassenhagen/
b1fca57f72e36b8c225d0315b18a0206

agramfort, 2017-07-14:
see https://gist.github.com/agramfort/c53483b8c94604ed464e002eba24f2b4

jona-sassenhagen, 2018-12-17:
I'm closing this, with metadata + examples in. Hope that's ok.

closed 2018-12-17 by jona-sassenhagen