mne-tools/mne-python #4126: Rearranging the tutorial + manual

choldgraf, 2017-03-29:
@kingjr and I have discussed a few times that the tutorial and manual are a bit noisy right now. I think it would help a lot if we could streamline these two pages on the website. Let's use this issue to discuss this in the hopes that I can put together something for tomorrow, and perhaps it can be merged by Friday.

As I see it, the Manual is for more "long form" guides, and the tutorial is for more "medium-form" guides (as opposed to the examples which are more "short-form" guides).

The problem is that right now there are long-form, textbook-like things in the tutorials (e.g., the signal processing tutorial), and there are more short-form MNE-like things in the manual (e.g., the MNE datasets manual page).

I am OK having a separate manual + tutorial, but I think that we should make sure that there is a clear and obvious delineation between them. We don't want users to be confused as to which of the two they should look into. Could we discuss this briefly tomorrow morning, and then I can open a PR to try a little rearranging?

I know that documentation isn't sexy but IMO it's one of the most important parts of the package :)

@kingjr @agramfort @Eric89GXL

jona-sassenhagen, 2017-03-30:
I think docs are very sexy actually! A lot of users are asking me about them, and I think they look pretty good already. (Not saying I disagree with your suggestion.)

Eric89GXL, 2017-03-30:
The distinction between manual/tutorial/examples is likely unclear from a user perspective.

The best thing might be to point out ideal examples people see in other packages and we can come up with concrete restructuring proposals from there.

kingjr, 2017-03-30:
IMO we should
1. fuse the tutorial and the manual, where we explain and motivate the
order of operations
2. simplify the examples as much as possible to really focus on *one*
function or object, i.e. force a < 30 lines of code per example.
3. increase the cross-referencing between examples, tutorials, docstrings

jona-sassenhagen, 2017-04-01:
Maybe we should think about a doc section discussing particular MNE conventions for which the API docs are insufficient. E.g., 'evend_id', 'events'. We get plenty of questions for both. E.g., ever since the introduction of "/"/HED-tags, `event_id` has become more interesting and nuanced.

dengemann, 2017-04-01:
I do not agree with everything here. But simplifying things is good. Keep
in mind our discussion about rearranging the columns and structuring the
tutorials page more like a text. See brainstorm/ fieldtrip. For examples I
would not impose a 30 lines limit but rather the constraint to show 1
figure / idea/ main output only.
I am not 100 percent convinced of merging tutorials and manual. But
renaming the manual to "background" or so and dividing content better
between the different doc sections could help a lot.

choldgraf, 2017-04-02:
I totally agree w/ @jona-sassenhagen about thinking of particular structures that aren't clearly documented but are often used (the events array is a good example of this). I'm taking a stab at doing a little housekeeping on the tutorials section and we can look at how this feels in an upcoming PR.

dengemann, 2017-04-02:
Going in smaller steps instead of monolithic helps. Be verbose. It is good
if we can join our attention on this one. Now after the sprint is a good
moment.

Eric89GXL, 2017-04-02:
I'm convinced we should merge the tutorials and manuals actually. I think
it would gain us a lot of clarity in presentation and access to information.

dengemann, 2017-04-02:
Then you need a good plan for that. My concern would be that it quickly be
too much and too dense. We would need a clear table of
contents/hierarchy/structure and a better overview page.

Eric89GXL, 2017-04-02:
Basically the flow diagram should be expanded, and the blocks of the
diagram provide the heading titles. Chris and I can try to start a PR to
show a proof of concept

dengemann, 2017-04-02:
Ok let's interact vis a PR. I will then setup builds and assist you.

PR #4173 (2017-05-02): MRG+3: Refactor documentation
This comes from a discussion with @choldgraf and @kingjr. It aims to simplify our docs so that users know exactly where to go when looking for information. It basically unifies `tutorials` and `manual` into a single being: `documentation`. I think for new users (and even established devs!) if we continue down this path of unification, it will be *much* easier for everyone to find information they need.

I think eventually we should also fold the `examples` into the `documentation`, as well, probably under some other subheading (advanced use cases or so?) within ea [... cut]
closed 2017-05-02 by agramfort