mne-tools/mne-python #4127: Writing a `to_bids` function

choldgraf, 2017-03-29:
@jasmainak you have a function to convert an MNE object to BIDS format, yes? I think it'd be a nice and relatively quick project to expose something like this that we could put into an example. WDYT? I'm happy to sit down and think through it if you like, or adapt your code to MNE.

teonbrooks, 2017-03-31:
- [ ] move it to the I/O module
- [ ] write to bids
- [ ] handle how MNE reads a bids folder structure
- [ ] finish the first round of this
- [ ] make a report for a BIDS directory

teonbrooks, 2017-03-31:
cc: @chrisfilo

teonbrooks, 2018-08-24:
closing since we now know have [mne-bids](https://github.com/mne-tools/mne-bids), which will be handling making files BIDS-compatible. inspired [here](https://github.com/mne-tools/mne-python/pull/4128#issuecomment-296484714)

closed 2018-08-24 by teonbrooks