mne-tools/mne-python #4129: Inconsistency within Dipole.

jaeilepp, 2017-03-30:
The ``Dipole`` constructor accepts only one dimensional ``times`` and ``amplitude``. ``_make_dipoles`` in ``_rap_music`` however accepts the whole time series worth of values for times and amplitudes. This causes the rap music example to fail on circle. I'm not sure which is the right way to fix this. Either way is an API change.

jaeilepp, 2017-03-30:
Or to be more specific, maybe the API doesn't need to change, but perhaps update the docs to mention that also the whole time series is accepted. I'm just not sure what to show for the dipole plotter with the full time series. The average amplitude and the whole time range?

agramfort, 2017-03-30:
for me Rap music gives the same thing as fitting a dipole at every instant
except dipole does not move. Just amplitude and eventually orientation does.

Eric89GXL, 2017-03-30:
It could / should probably return DipoleFixed in that case

Eric89GXL, 2017-04-18:
Fixed by @jaeilepp's PR

closed 2017-04-18 by Eric89GXL