mne-tools/mne-python #4130: Forward model tutorial not working (sample data problem?)

er06645810, 2017-03-30:
I'm trying to go through the tutorial on creating a forward solution [here](http://martinos.org/mne/stable/auto_tutorials/plot_forward.html). I'm behind a proxy, so I wasn't able to download the data automatically. Instead I snagged the URL out of the source code and downloaded it manually. I then decompressed the folder and stuck it in my working directory.

The only change I made to the tutorial code was to specify
```python
data_path = sample.data_path('./')
```
instead of using the default environment variables, since I want to keep the data with my script.

Following the tutorial exactly, I get an error at line 439 of `surface.py` in `_fread3(fobj)`. The error is 
```python
ValueError: not enough values to unpack (expected 3, got 0)
```
The same error occurs both times that `mne.viz.plot_bem()` is called in the tutorial.

Can I safely ignore this error and move on with creating the forward solution, or is the fact that I can't plot the data indicative of a problem that is going to ruin the foward model?

jaeilepp, 2017-03-30:
Can you provide the full stack trace?

er06645810, 2017-03-30:
Something along these lines

```
Using surface: C:\Users\SHIUser\sumi-forward-model\MNE-sample-data/subjects\sample\bem\inner_skull.surf
Using surface: C:\Users\SHIUser\sumi-forward-model\MNE-sample-data/subjects\sample\bem\outer_skull.surf
Using surface: C:\Users\SHIUser\sumi-forward-model\MNE-sample-data/subjects\sample\bem\outer_skin.surf

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-64-7847f8865159> in <module>()
----> 1 mne.viz.plot_bem(subject, subjects_dir, brain_surfaces='white', src=src, orientation='coronal')

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\viz\misc.py in plot_bem(subject, subjects_dir, orientation, slices, brain_surfaces, src, show)
    443     # Plot the contours
    444     return _plot_mri_contours(mri_fname, surfaces, src, orientation, slices,
--> 445                               show)
    446 
    447 

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\viz\misc.py in _plot_mri_contours(mri_fname, surfaces, src, orientation, slices, show)
    309     for file_name, color in surfaces:
    310         surf = dict()
--> 311         surf['rr'], surf['tris'] = read_surface(file_name)
    312         # move back surface to MRI coordinate system
    313         surf['rr'] = nib.affines.apply_affine(trans, surf['rr'])

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\surface.py in read_surface(fname, read_metadata, return_dict, verbose)

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\utils.py in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\surface.py in read_surface(fname, read_metadata, return_dict, verbose)
    510     read_tri
    511     """
--> 512     ret = _get_read_geometry()(fname, read_metadata=read_metadata)
    513     if return_dict:
    514         ret += (dict(rr=ret[0], tris=ret[1], ntri=len(ret[1]), use_tris=ret[1],

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\fixes.py in _read_geometry(filepath, read_metadata, read_stamp)
     98     NEW_QUAD_MAGIC = 16777213
     99     with open(filepath, "rb") as fobj:
--> 100         magic = _fread3(fobj)
    101         if magic in (QUAD_MAGIC, NEW_QUAD_MAGIC):  # Quad file
    102             nvert = _fread3(fobj)

C:\Users\SHIUser\Anaconda3\lib\site-packages\mne\surface.py in _fread3(fobj)
    437 def _fread3(fobj):
    438     """Read 3 bytes and adjust."""
--> 439     b1, b2, b3 = np.fromfile(fobj, ">u1", 3)
    440     return (b1 << 16) + (b2 << 8) + b3
    441 

ValueError: not enough values to unpack (expected 3, got 0)
```

jaeilepp, 2017-03-31:
That would suggest something is wrong with the surface files. Can you check if ``inner_skull.surf``, ``outer_skull.surf`` and ``outer_skin.surf`` are present and readable? They should be in the ``bem`` folder of ``sample`` subject.

er06645810, 2017-04-03:
Thanks for the guidance. All three files are present, but have a file sizeo of 0 KB. I'll try to download the sample dataset again and let you know if that fixes the problem.

Eric89GXL, 2017-04-03:
Closing for now, then.

Are you on Windows? I wonder if those are actually symbolic links that don't get resolved properly.

closed 2017-04-03 by Eric89GXL

er06645810, 2017-04-04:
@Eric89GXL Your intuition was spot on.
[image]

Windows 8, 7zip.

er06645810, 2017-04-04:
Opening 7zip as administrator and rerunning the extraction solved the problem. Thanks for the guidance.