mne-tools/mne-python #4131: use SNR=1 for apply_inverse_epochs

agramfort, 2017-03-30:
see

http://martinos.org/mne/stable/auto_examples/inverse/plot_compute_mne_inverse_epochs_in_label.html

snr = 1 should be used and explained

haribharadwaj, 2017-04-02:
I disagree, as I mentioned in the e-mail list:
"Yes, it's true of course that single trials are noisier, but often single trial  processing in the source space is followed by combining across trials.. So in many (but not all) instances, it is useful to have a consistent inverse operator (specifying a fixed SNR... and for dSPM scaling the noise cov by a fixed number of trials).

For instance, if you did inverse first on single trials and then averaging in source space vs. average across trials in sensor space first and then do inverse, you would get different answers if you specified different SNRs for the two cases.

This issue came up before, and the 3.0 was used to make them consistent:

https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2014-April/002066.html

https://github.com/mne-tools/mne-python/pull/1237

Just something to keep in mind depending on how you are using single trial inverses."

agramfort, 2017-04-03:
ok let's add comments in the examples to explain this then

dengemann, 2017-04-03:
Another good GSOC PR :)

closed 2018-03-26 by agramfort