mne-tools/mne-python #4133: have rap music return a DipoleFixed

agramfort, 2017-03-30:
rap_music should return DipoleFixed not Dipole

maybe a good issue for a GSOC candidate?

rodrigohubner, 2017-03-31:
I will try to reproduce it. If I find the error in question I'll try to fix it right away.

agramfort, 2017-03-31:
go for it !

rodrigohubner, 2017-03-31:
I managed to reproduce and really `rap_music` returns `Dipole` instead of `DipoleFixed`. OK!
Should the `rap_music` function always return `DipoleFixed`? If so, I'll change a lot in the _rap_music.py file from the Dipole class instance to DipoleFixed. Was it correct?

agramfort, 2017-03-31:
correct !

rodrigohubner, 2017-03-31:
I made a PR for it.
https://github.com/mne-tools/mne-python/pull/4148

joewalter, 2017-04-01:
IIRC our plotting tools for dipoles, including the new orthoview, expect Dipole objects, not DipoleFixed. So before, we apply this big change here, it should be made sure that there are the same options for visualization for DipoleFixed. So I am -1 if only the rap-music code and tests are adapted without having the same visualization tools as before.

agramfort, 2017-04-01:
agreed. Let's update the plot to support DipoleFixed too.

Eric89GXL, 2017-04-02:
Yeah I think there is not `ori` parameter for `DipoleFixed` :(

agramfort, 2017-04-03:
hum :( what's the way out?

joewalter, 2017-04-03:
So why don't we just leave it as a Dipole object (and adapt the doc of Dipole)? DipoleFixed seems to be used for dipole fiiting only for fixed position and orientation. So it might even work for rap music, but is not suitable for sparse imaging methods.

Eric89GXL, 2017-04-03:
> DipoleFixed seems to be used for dipole fiiting only for fixed position and orientation.

Fixed position is the only requirement. The orientation can change. I think it just doesn't get saved anywhere. But even so, I agree `Dipole` is probably a better way to go because it gives us the orientations.

Eric89GXL, 2017-04-18:
I think we've dropped this idea

closed 2017-04-18 by Eric89GXL