mne-tools/mne-python #4150: jupyter 3D plotting

choldgraf, 2017-04-02:
This is an issue to carry on conversation / development of incorporating pythreejs into some plotting functionality. It sounds like we've already got a pretty good product right now, but I want to make sure that we get the last 20-30% of the way there.

@agramfort do you have plans to merge in anything you were working on at the sprint? or still need more development time?

maybe @Eric89GXL has thoughts too, as I could see this being useful for the pysurfer world too...maybe it's worth pinging them about this.

Eric89GXL, 2017-04-02:
> maybe @Eric89GXL has thoughts too, as I could see this being useful for the pysurfer world too...maybe it's worth pinging them about this.

Yeah one possible test would be to ship it as a replacement backend for PySurfer. If it works there then people's workflows can be easily covered.

However, breaking from PySurfer also has some potential advantages. So who knows.

agramfort, 2017-04-03:
we'll keep the ball rolling over the next months to see how much we can
push in this direction.

choldgraf, 2017-04-04:
Great - do we have any kind of PR open? Or was all the stuff we demoed last week just pilot material on your computer?

choldgraf, 2017-04-04:
(in which case, it'd be useful to link to a gist or a binder link so people can demo the functionality, I'm happy to set this up if you point me to the code)

agramfort, 2017-04-04:
here was my starting gist:

https://gist.github.com/agramfort/3ebd5063b24f11eccd0a43e50a8ab348

further improved by @jason_grout

jasongrout, 2017-04-12:
Here's my gist: https://gist.github.com/jasongrout/2ae29e9d2a84c4f06cb664e5df229273

choldgraf, 2017-04-12:
I'm getting this error:

```
/Users/choldgraf/anaconda/lib/python3.6/site-packages/ipykernel/jsonutil.py in json_clean(obj)
    171 
    172     # we don't understand it, it's probably an unserializable object
--> 173     raise ValueError("Can't clean for JSON: %r" % obj)

ValueError: Can't clean for JSON: <memory at 0x10955c558>
```
I pip-installed pythreejs from git, so I should be working from master. Any idea what's up? Is this the problem we were discussing about pythreejs not knowing what to do with numpy arrays? I thought that was fixed...

jasongrout, 2017-04-12:
ipywidgets and widgetsnbextension also need to installed from master.

choldgraf, 2017-04-12:
if you install ipywidgets from master do you also get widgetsnbextension? I couldn't find a repo for it

choldgraf, 2017-04-12:
ah I see it has its own `setup.py`...so I assume I need to install manually

choldgraf, 2017-04-12:
hmmm, am having a lot of issues getting master branches to build. I'll try and debug this later.

choldgraf, 2017-05-01:
wew - got it working!

dengemann, 2017-05-01:
Wow! Video/images?

choldgraf, 2017-05-01:
first I need to figure out how to do it without overloading my computer's memory :) I'll push something to a little demo repo in a bit tho. Right now it's basically what @jasongrout had tho

choldgraf, 2017-05-02:
@jasongrout for what it's worth, it seems like the memory / slowdown problems happen when you connect the slider to the 3D viz...any reason you can think of that the interactive slider stuff would slow things down?

jasongrout, 2017-05-02:
I wonder if there is memory leak. Can you just change things in a for loop and see if that also slows things down?

choldgraf, 2017-05-02:
hmmm, it does seem to bog down when I do a for loop changing the color each time...here's my memory over time during a for loop. You can tell when the for loop starts ;-)

[image]

also unrelated - did you ever figure out how to get the xyz rotation working properly? I can't get the brain to have the proper side view...

choldgraf, 2017-05-02:
and once I reload the page (on the right):
 
[image]

choldgraf, 2017-05-02:
more thoughts: this doesn't seem to happen if I re-run the same loop over and over again, if the loop is small. However if the loop is larger (e.g., update the plot 20 times instead of 4 times) then it seems to start accumulating memory...

larsoner, 2019-08-20:
Closing for @GuillaumeFavelier's work

closed 2019-08-20 by larsoner