mne-tools/mne-python #4157: "persistent" info per epoch?

timmidee, 2017-04-03:
Is there a way to keep information with each epoch in an instance of epochs? (e.g an RT, a stimulus identifier, a trial ID, or any kind of info, really). Preferably it would be some way that is also robust to operations like AR.

E.g. I want to plot an ERP image, sorted by the value of some variable that I have per epoch. If I keep the values per trial/epoch externally, the values do not line up with the data after AR or similar operations.

agramfort, 2017-04-04:
we discussed last week about this use case that pops up often.

I think we need a dataset to illustrate this feature and see what's the
best way to proceed.

do we have a good public dataset that would motivate this feature?

jona-sassenhagen, 2017-04-04:
@agramfort I have a few. I can contribute a linguistic one with a lot of vars, or simply one with a complex design + RTs.

jona-sassenhagen, 2017-04-04:
What's the constraints for datasets? (E.g., size, has to be published ...)

jona-sassenhagen, 2017-04-04:
This would also be good for XDawn tests/docs btw.

jona-sassenhagen, 2017-04-04:
@timmidee or we could even use one of yours, a free-reading one ...

timmidee, 2017-04-04:
Good to know this use case is on the radar. I can see if I can contribute some data with eye movements, but that would pose an additional possible problem of overlapping events. Perhaps that's less than ideal if the goal is to solve this specific issue?

agramfort, 2017-04-04:
whatever works guys. It has to be public. It's better if it's published.

timmidee, 2017-04-24:
@jona-sassenhagen the datafile I sent you for #4213 contains eye movement info with a relatively high number of parameters per event. If we just drop the overlapping events, trim the length and downsample (and accept the events dropped in that) it might work as an example for this?

I might want to change/delete some fields or field names.

It's "public" in a sense that with a few minor edits I have no problems with it freely circulating.
I don't have any published data that could be helpful here.

jona-sassenhagen, 2018-03-31:
This is solved by metadata, correct?

closed 2018-03-31 by jona-sassenhagen