mne-tools/mne-python #4159: MINOR: use bootstrap for GFP CI for plot_compare_evokeds

jona-sassenhagen, 2017-04-04:
`plot_compare_evokeds` will calculate an analytic CI for channels, but as the GFP isn't normally distributed, I decided against calculating the CI. (Actually, isn't grad RMS also not normally distributed at all ..?)
@dengemann just contributed a helper function for bootstrapped CIs for GFP, which should be added to the `plot_compare_evokeds` code.

Maybe it should always be used? Or maybe just for GFP and RMS.

dengemann, 2017-04-04:
Jona can you play with it a bit? We might want to generalize it and add it
to mne.stats.

mmagnuski, 2017-04-06:
For highly skewed distributions one way would be to fit gamma distribution and estimate ci from gamma. A simpler approach is to calculate one-sided sd's (this is what is being done in ica's plot_properties [here](https://github.com/mne-tools/mne-python/blob/master/mne/viz/ica.py#L271). But bootstrapping is also good if it's fast - it would be good to have it in stats.

jona-sassenhagen, 2017-04-06:
Hm. I'm okay with either. @dengemann , opinions?

I guess the bootstrap is more robust, the gamma *might* be more effective and efficient.

We could easily give the option to pass a function to `ci`, but it should default to something sensible otherwise. (The question is, should it pick gamma/bootstrap only for RMS/GFP, or should it do some basic check (are all values positive?), or should it always bootstrap, ..?)

mmagnuski, 2017-04-06:
Bootstrap or one-sided std are safer, scipys gamma sometimes does bad fits unless loc is frozen or good starting parameters are given. And while bootstrap and one-sided sd will work both on gaussian and skewed, gamma requires the distribution to be skewed (although it approaches gaussian with increasing shape param).

dengemann, 2017-04-06:
Ok you convince me to write tests and open a pr for a generalized bootstrap
function. Bootstrap is cool as it makes almost no assumptions and can be
used with any statistic.

jona-sassenhagen, 2017-04-06:
Do we have any synergy with the other stats functions? (Which also use resampling.) Maybe we can keep actual overhead low.

dengemann, 2017-04-06:
I think this is pretty independent and adds no overhead.

dengemann, 2017-04-06:
That's why you find it in an example :)

jona-sassenhagen, 2017-09-06:
Done in #4289

closed 2017-09-06 by jona-sassenhagen