mne-tools/mne-python #4175: raw.crop(): raw.first_samp unchanged

DominiqueMakowski, 2017-04-07:
Hey,
I cropped my raw eeg data. Now the raw.times index is correct (*i.e.,* starts from 0). However, when I do `print(raw.first_samp)` it returns the tmin value with which I cropped the data... This prevents me from adding events, as it says `ValueError: event sample numbers must be between 58052 and 358101` (which are the cropping indexes).

Is this behaviour normal? Is there a way to reset the first_samp values?
thanks

agramfort, 2017-04-08:
can you share a code snippet on sample to better understand?

DominiqueMakowski, 2017-04-14:
I've got [here](https://github.com/DominiqueMakowski/MakTools/tree/master/eeg_test) a code snippet and the data file example that I use to illustrate the crop-related issue. 

However, I've run into another problem. I can't apply ICA for ECG artifacts correction: it throws a `memory error`. The thing is, I've tried this code snippet on two beefy computers (16go RAM, i7 and so on) and it returns the same error. Am I doing something wrong? Thanks!

Eric89GXL, 2017-04-14:
Can you paste the traceback for the MemoryError? It will give some insight into the failure.

Does your memory almost run out before the error? You can use your system monitor of choice to eyeball it during the failure.

DominiqueMakowski, 2017-04-14:
Here's the traceback:

```python
Opening raw data file E:\Dropbox\RECHERCHE\Misc\MakTools\eeg_test\example_raw.fif...
Isotrak not found
    Range : 0 ... 366159 =      0.000 ...   366.159 secs
Ready.
Reading 0 ... 366159  =      0.000 ...   366.159 secs...
51522
Setting up band-pass filter from 1 - 40 Hz
l_trans_bandwidth chosen to be 1.0 Hz
h_trans_bandwidth chosen to be 10.0 Hz
Filter length of 6600 samples (6.600 sec) selected
Fitting ICA to data using 63 channels. 
Please be patient, this may take some time
Inferring max_pca_components from picks.
Selection by number: 25 components
C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\sklearn\decomposition\fastica_.py:116: UserWarning: FastICA did not converge. Consider increasing tolerance or the maximum number of iterations.
  warnings.warn('FastICA did not converge. Consider increasing '
Using channel ECG to identify heart beats.
Setting up band-pass filter from 8 - 16 Hz
Number of ECG events detected : 382 (average pulse 76 / min.)
382 matching events found
Loading data for 382 events and 1001 original time points ...
0 bad epochs dropped
Traceback (most recent call last):

  File "<ipython-input-2-6a52256cec99>", line 1, in <module>
    runfile('E:/Dropbox/RECHERCHE/Misc/MakTools/eeg_test/test.py', wdir='E:/Dropbox/RECHERCHE/Misc/MakTools/eeg_test')

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\spyder\utils\site\sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "E:/Dropbox/RECHERCHE/Misc/MakTools/eeg_test/test.py", line 33, in <module>
    ecg_inds, scores = ica.find_bads_ecg(ecg_epochs, method='ctps')

  File "<string>", line 2, in find_bads_ecg

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\mne\utils.py", line 707, in verbose
    return function(*args, **kwargs)

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\mne\preprocessing\ica.py", line 1036, in find_bads_ecg
    sources = self.get_sources(inst).get_data()

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\mne\preprocessing\ica.py", line 726, in get_sources
    sources = self._sources_as_epochs(inst, add_channels, False)

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\mne\preprocessing\ica.py", line 776, in _sources_as_epochs
    sources = self._transform_epochs(epochs, concatenate)

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\mne\preprocessing\ica.py", line 652, in _transform_epochs
    data = np.hstack(epochs.get_data()[:, picks])

  File "C:\Users\DomM\Desktop\WinPython-32bit-3.5.3.0Qt5\python-3.5.3\lib\site-packages\numpy\core\shape_base.py", line 288, in hstack
    return _nx.concatenate(arrs, 1)

MemoryError
```

Also, here's the evolution of memory and processor during the failure. The failure happened when the processor signal drops down, which corrresponds to a small wave in the memory signal...

[image]

Eric89GXL, 2017-04-14:
The `hstack` of EEG data is a potentially big copy, so it is probably a real failure.

You can use `decim` to work around this. It will also speed up your ICA. It only uses every Nth sample, but it seems like you will still have plenty of samples to fit the model. But @agramfort or @dengemann probably has more insight on that.

dengemann, 2017-04-14:
We optimised memory in ICA down to scikit-learn. As long as that code did not change there won't be too much left to optimize. As Eric said you do not need all the samples to fit a good ICA. This of course depends on what your goal is. @DominiqueMakowski can you tell me which line your code is running into memory problems? Which function call is it?

Eric89GXL, 2017-04-14:
@dengemann see stack trace above

DominiqueMakowski, 2017-04-15:
What a dumbass I am I was using a 32bit python version 😬 Sorry!!

closed 2017-04-15 by DominiqueMakowski