mne-tools/mne-python #4178: ica.save() method failure when passing a variable for n_components

craddm, 2017-04-10:
Hi,

Here's a slightly niche one: after running ICA, I tried to save the ICA object using its .save() method. But it failed with this message: 

PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'test-ica.fif'

I figured out that this is due to the way I check the rank of the data before running ICA and pass the result of the rank check to the ICA function. 

rank = np.linalg.matrix_rank(np.squeeze(epochs[0].get_data()))
method = 'extended-infomax'
ica = ICA(method = method, n_components = rank)

Obviously I want to do that to be able to run ICA in a batch script  - the results can be nonsense if the data is not full rank and no PCA is done, and since my data is average referenced EEG, the data is never full rank. The problem is rank is a Numpy int32, and cannot be serialized. I'm on Python 3.6 - I know you've built some checks in that catch this error in some circumstances, but this seems to skip them. It's easy to fix by forcing it to a normal int before passing it to ICA, but just thought you might want to know about it.

Cheers,
Matt

jaeilepp, 2017-04-10:
I'm unable to reproduce this, but I'm on Linux. Is it so that if you cast ``rank`` to int before calling ``ICA``, it works? That would be a bit weird since that's done already at line [1745](https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/ica.py#L1745). What is your version of MNE?

jaeilepp, 2017-04-10:
Or actually, it's probably not done to it since the conditional fails on numpy int. But still, can you confirm that casting to int fixes it?

craddm, 2017-04-10:
Yes, `int(rank)` fixes it. (MNE version is 0.14)

agramfort, 2017-04-10:
can you send a PR that does the cast to int when writing or directly in the
constructor?

jaeilepp, 2017-04-10:
@agramfort see https://github.com/mne-tools/mne-python/pull/4179

closed 2017-04-11 by agramfort