mne-tools/mne-python #4189: How to create raw and then epochs from self data array

thisroot, 2017-04-12:
Hello, for several days I can not solve the problem. I have an array (x,y). Now I create a raw object who contain items of my array.

```np.where(raw_train == array[0][0][0]) => true ```

then i create events array =>
```
[
[0,0,2]
[4943,0,1]
[9325,0,1]
...
]
```
And  make epochs object

```
epochs = mne.Epochs(raw_mne,
                                     events,
                                     event_id=labels,
                                     add_eeg_ref=False, 
                                     preload = True, 
                                     tmin = 0, tmax = 4.499)
```

There has several questions:
events contains not cutting ranges, and i don't know these ranges. I set up Tmax variable and to get epochs. 
does correctly I realized that the ends of each activity are cropped until 4500 miliseconds?

But now I can not find the value of my original array in an array of epochs

```
data_epochs = epochs.get_data()
np.where(data_epochs == array[0][0][0]) => False
```

I can not understand what I'm doing wrong

jaeilepp, 2017-04-13:
Can you provide a runnable piece of code that demonstrates this problem.

agramfort, 2017-07-21:
closing due to no answer

closed 2017-07-21 by agramfort