mne-tools/mne-python #4190: raw.plot() silently applies projectors that are disabled

drammock, 2017-04-12:
**how it works now**

- inactive projectors are automatically applied during `raw.plot()`
- a small "proj" button appears in the bottom right of the plot window
- Clicking the proj button shows a list of projectors, each with a large "X" next to it.  

**how it should work**

- if a raw object has projectors that are ***not activated***, `raw.plot()` should plot the ***untransformed data***
- the big "X" marks should be replaced with check marks, or something else that is less ambiguous (maybe click on the projector names to make them black vs. grey, analogous to bad channels?).  It is quite plausible to interpret "X" as either "selected" (i.e., "active") or as "not active".
- at a minimum, the whole phenomenon of the "proj" button should be documented in the `raw.plot()` docstring.
- optionally, `raw.plot()` could get a `proj` argument that lets the projectors get applied (non-interactively) before plotting.  The simplest version might accept only "all or none" booleans; a more complex implementation might accept a list of indices into `raw.projs` indicating which projections to activate.

Eric89GXL, 2017-04-12:
"active" means "have been applied to the data". But it's a bit of a misnomer as when you do source localization, both active and inactive end up being used/applied to the data.

> if a raw object has projectors that are not activated, raw.plot() should plot the untransformed data

For backward compat (and less importantly, least surprise for mne_browse converts) I don't think we should change this. The standard MNE pipeline will eventually apply them (`proj=True` in several places) so there is some rationale for it.

But +1 for visual clarifications, doc improvements, and even a `proj=True (default) | False` option.

drammock, 2017-04-12:
yeah, I know what "active" means.  That's not the problem.  There is advice in the docs that says (I'm paraphrasing):

> avoid applying projectors too early; use `<obj>.copy().apply_proj().plot()` to see what they're doing without altering the original data.  

The (undocumented) interactive "proj" button makes that advice moot (you don't need to make a copy first in order to see what changes a projector is making to the signal).  Even more perversely, if you do follow the advice, it doesn't actually work: both plots look the same, i.e., you get identical plots from `raw` and `raw2`:
```
raw.add_proj(blink_proj)  # or whatever other projector you want
raw2 = raw.copy()
raw2.apply_proj()
raw.plot()
raw2.plot()
```
If preserving backward compatibility is that important in this case, I think having an explicit `proj=True` as part of the function signature is necessary, to draw attention to the fact that this is happening by default.

agramfort, 2017-04-13:
I'd be fine with

raw.plot(proj=False)

option

Eric89GXL, 2017-04-18:
@drammock do you have time to make a `proj=True` default PR?

drammock, 2017-04-18:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

I have a presentation late this afternoon, planning to start on it
tomorrow.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJY9jDkAAoJEEzlbMQqUVLONtcH/3eP6/L86AM1Gp6sye2IkIn/
yp0fR2O6DTAsPa4cdXIeBI1YH8xCV8KIDA9dvKWsIFbUEh1aq1NHGCh/gourjbz3
K7k+oRsfEp3S5IG3BN6+Qn8D2DReykFpTsr61d31/fqQwf5BuD5XYb1STVcXgoP0
jErMcj9IF4zdJEZbQTW7Sgr93x6FRjx7hvLHcY04FzxcMz5C84irlUow3qb/Bl28
+I3vsZ/pPurWXxhD/IOtOktwFeJiElLiKHc4oRrWAZ7m068+H2Mk84P88bu5hk3K
/x/+nVg2+Vke6lkMZvOyKjUe8MuwASvqYfB/xB7qkoZqrtlNxZFNA1UhVC8+5XQ=
=mCZs
-----END PGP SIGNATURE-----

PR #4210 (2017-04-21): adds proj=True argument to plot_raw
closes #4190.  @Eric89GXL are you reviewing this one or should I ping someone else?
closed 2017-04-21 by Eric89GXL