mne-tools/mne-python #4192: BUG?: environment variables for testing dataset aren't respected?

bloyl, 2017-04-14:
When I set the environment variable 'MNE_DATASETS_TESTING_PATH' I don't get a data_path.
```
In [1]: import os
In [2]: import mne
In [3]: print [k for k in os.environ if k.upper().startswith('MNE')]
[]
In [4]: mne.datasets.testing.data_path(download=False)
Out[4]: u'/home/bloyl/mne_data/MNE-testing-data'
In [5]: ls /home/bloyl/mne_data/MNE-testing-data
ARTEMIS123/  BTi/  CNT/  CTF/  datasets/  dip/  EDF/  EEGLAB/  MEG/  misc/  montage/  README.md  SSS/  subjects/  version.txt
In [6]: os.environ['MNE_DATASETS_TESTING_PATH'] = '/home/bloyl/mne_data/MNE-testing-data'
In [7]: mne.datasets.testing.data_path(download=False)
Out[7]: ''
```
I haven't tried any of the other datasets but i imagine something similar will happen. Is there another way to set the path that I should use?

Eric89GXL, 2017-04-14:
I can't reproduce:
```
omar:~ larsoner$ MNE_BLAH=foo python -c "import os; print([k for k in os.environ if k.upper().startswith('MNE')])"
['MNE_BLAH']
```
or
```
omar:~ larsoner$ export MNE_BLAH=bar
omar:~ larsoner$ python -c "import os; print([(k, v) for k, v in os.environ.items() if k.upper().startswith('MNE')])"
[('MNE_BLAH', 'bar')]
```
But really the recommended permanent way to set it (in the MNE config file) is:
```
mne.set_config(...)
```

Eric89GXL, 2017-04-14:
Oh wait you set `os.environ` directly... let me look again.

Eric89GXL, 2017-04-14:
Yeah I can reproduce it -- even if I set it to the value in the MNE config it doesn't work. We must have some bug.

bloyl, 2018-04-05:
So to be honest, I"m not sure how/why I thought it should work. maybe just looking through code.

However it might make sense to document the environment variables for the datasets somewhere
maybe https://martinos.org/mne/stable/manual/datasets_index.html#datasets?

reason being is that in multi user environments its nice to have a single copy of those datasets. Is there a way to specify that location globally (for all users) using an mne config file or something?

larsoner, 2018-04-05:
No global location. You could provide a template `~/.mne/mne-python.json` file though.

PR #5101 (2018-04-05): MRG: Add test
Closes #4192.

Not really a critical test but it takes almost no time, so why not.
closed 2018-04-05 by agramfort