mne-tools/mne-python #4194: other datastores

howarth, 2017-04-14:
Hello, my name is Dan Howarth and I work in Tom Mitchell's brain lab. Our lab is putting all of our MEG data into a distributed store. It would be nice if we could load data into a Raw, Epochs, Evoked. I'm envisioning something like

raw = BuildRaw(infoobject, data, time)
Where data is a numpy array of shape (n channels , n time points) and time is (1, time points)

epochs = BuildEpochs(infoobject, data, time) 
evoked = BuildEvoked(infoobject, data, time)
where data is a numpy of shape (n segments, n channels, n time points)

If it makes it easier infoobject could be a raw object without any loaded data.

I looked into the source code a bit to see if I could just switch out the "read segment" function to have it read from my store, but it seemed like that was too entangled with other parts of the object. 

I'd like to add this feature, but need some guidance on how I could build such a device. I have a degree is CS and am ok with figuring this out on my own, but I'm just hoping that maybe one of the developers is aware of a clear path to do that.

jaeilepp, 2017-04-14:
Have you checked out our [array constructors](https://mne-tools.github.io/dev/python_reference.html#creating-data-objects-from-arrays)? It sounds like they're just what you need. There's also this [tutorial](https://mne-tools.github.io/dev/auto_tutorials/plot_creating_data_structures.html). Let me know if those don't do the job for you.

howarth, 2017-04-14:
Thanks! A quick scan of that looks like it's exactly what I wanted

agramfort, 2017-04-14:
out of curiosity what data store are you using? something custom?

howarth, 2017-04-14:
Right now Cassandra is the underlying store but we're designing a layer on top of that so other things can easily be switched out.

dengemann, 2017-04-14:
Hi Dan!
I am not sure I understand. Did you look at mne.RawArray, mne.EpochsArray
et al?
http://martinos.org/mne/dev/auto_examples/io/plot_objects_from_arrays.html

Denis

dengemann, 2017-04-14:
Oops sorry did not see Jaakko's reply.

agramfort, 2017-04-14:
>
> Right now Cassandra is the underlying store but we're designing a layer on
> top of that so other things can easily be switched out.

I'd be really curious to hear about your experience with this technology.
come say hi if you see me in a conf or maybe write a blog post :)

Eric89GXL, 2017-04-18:
Sounds like we're good here

closed 2017-04-18 by Eric89GXL