mne-tools/mne-python #4198: Can not load '.BDF' file

waa8128, 2017-04-16:
I wanted to load a '.bdf' file but failed.

```
raw = mne.io.read_raw_edf('sub01_block1.bdf',  stim_channel = -1)

Extracting edf Parameters from sub01_block1.bdf...
Setting channel info structure...
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-39-4b97df540fa5> in <module>()
----> 1 raw = mne.io.read_raw_edf('sub01_block1.bdf',  stim_channel = -1)

D:\Anaconda3\lib\site-packages\mne\io\edf\edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    606     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
    607                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
--> 608                   exclude=exclude, preload=preload, verbose=verbose)

D:\Anaconda3\lib\site-packages\mne\io\edf\edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

D:\Anaconda3\lib\site-packages\mne\utils.py in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

D:\Anaconda3\lib\site-packages\mne\io\edf\edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
     77         input_fname = os.path.abspath(input_fname)
     78         info, edf_info = _get_edf_info(input_fname, stim_channel, annot,
---> 79                                        annotmap, eog, misc, exclude, preload)
     80         logger.info('Creating Raw.info structure...')
     81         _check_update_montage(info, montage)

D:\Anaconda3\lib\site-packages\mne\io\edf\edf.py in _get_edf_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    481             info['highpass'] = float(highpass[0])
    482     else:
--> 483         info['highpass'] = float(np.max(highpass))
    484         warn('Channels contain different highpass filters. Highest filter '
    485              'setting will be stored.')

TypeError: float() argument must be a string or a number, not 'list'
```

​

agramfort, 2017-04-16:
can you share the file so we can investigate?

thx

agramfort, 2017-07-21:
closing due to no answer

closed 2017-07-21 by agramfort