mne-tools/mne-python #4200: How did we set up the auto-gallery for API pages?

choldgraf, 2017-04-17:
I think either @Eric89GXL or @agramfort did this, so pinging them...

I'm trying to help Nelle migrate Matplotlib to sphinx-gallery. One thing we'd like to do is get the API docs to start including sphinx gallery thumbnails automatically, kinda like this page:

http://martinos.org/mne/stable/generated/mne.read_epochs.html#mne.read_epochs

Who set this up for MNE? I'd love a quick runthrough of how to do this if it's possible...

agramfort, 2017-04-17:
everything is there:

https://github.com/mne-tools/mne-python/blob/master/doc/conf.py#L301

there was not much magic / hacking to make it work

dengemann, 2017-04-17:
Also see mne-hcp repo where these conf params are adjusted.

choldgraf, 2017-04-17:
What is the purpose of `'mod_example_dir': 'generated'`? I can't see any mention of it on the sphinx gallery website but it's in both of these projects

Eric89GXL, 2017-04-17:
It has been renamed to `backreferences_dir` and we are just behind the times

choldgraf, 2017-04-17:
ahh I see - ok so both will work, that's good to know. I'm gonna leave this open for a little bit because I may wanna ping you guys about this but I'll close it soon

choldgraf, 2017-04-17:
(unless you'd rather talk offline about it, but I figured it's useful to have up here for posterity)

Eric89GXL, 2017-04-18:
We can continue to talk here as necessary but I'll close it so it doesn't show up as an open issue

closed 2017-04-18 by Eric89GXL