mne-tools/mne-python #4201: ch_names/montage incompatibility with GSN-HydroCel (.edf files)

NicolasLgrnd, 2017-04-18:
I tried to import a '.edf' with mne.io.read_raw_edf but it seems that I encounter a problem already described [here](https://github.com/mne-tools/mne-python/issues/4076) : ch_names are listed as 'EEG 1', 'EEG 2'.... while in the GSN-HydroCel montage files it is in the form 'E1', 'E2' ...

` ValueError: None of the sensors defined in the montage were found in the info structure. Check the channel names.`

Thanks in advance for your help

agramfort, 2017-04-18:
it was fixed in:

https://github.com/mne-tools/mne-python/pull/4079

it might not be in the release though.

See channel_renaming parameter

please close the issue if you're good

jaeilepp, 2017-04-18:
That's for egi files. Perhaps this needs to be unified across readers.

NicolasLgrnd, 2017-04-18:
I tried with v 0.15 but it didn't work neither, maybe the bug described in #4079 needs to be fixed for .edf files too.

Renaming channels does the job.

I close here,

Thanks for your help

closed 2017-04-18 by NicolasLgrnd

jaeilepp, 2017-04-18:
I looked at the edf code, and it seems that the channel names are actually read from the header. So maybe the cleanest solution is just to make your own copy of the montage file where the channel names match the data. It would of course be possible to add a param for the ``read_montage`` to allow mapping data channels to montage channels, but I don't think it's worth the effort. Basically it would require the same effort for the user as changing the montage file directly.