mne-tools/mne-python #4206: mne.viz.plot_trans failed

romquentin, 2017-04-18:
Hi mne-python experts

I have an issue running:

`plot_trans(epochs.info, trans=None, subject=freesurf_subject, dig=True,
           meg_sensors=True, subjects_dir=subjects_dir, brain=True)`

it gives me this error:

/Applications/anaconda/lib/python2.7/site-packages/mne-python/mne/viz/_3d.pyc in _fiducial_coords(points, coord_frame)
     52     points_ = dict((p['ident'], p) for p in points if
     53                    p['kind'] == FIFF.FIFFV_POINT_CARDINAL)
---> 54     return np.array([points_[i]['r'] for i in FIDUCIAL_ORDER])
     55 
     56 

KeyError: 1

I am running on macOS X El capitan and with mne.0.15dev.
The error seems to be not present on mne.0.13stable.

Thank you

Romain

Eric89GXL, 2017-04-18:
We should add a check for this. 

But you have strange data -- you have digitization points but no fiducials?

romquentin, 2017-04-18:
I have 3 digitization points corresponding to the fiducials in surface_MRI coordinate system.

Eric89GXL, 2017-04-18:
Ahh yes, this problem again :) Okay we can make `plot_trans` work in the short term and make it actually plot those properly in the long term.

romquentin, 2017-04-18:
Yes :)
Ok. Thank you!

Eric89GXL, 2017-04-18:
Can you try #4069 to see if it fixes your problem?

romquentin, 2017-04-18:
It fixed it! Thank you!

Eric89GXL, 2017-04-20:
Should be fixed in `master` now

closed 2017-04-20 by Eric89GXL