mne-tools/mne-python #4207: ENH: streamline processing CTF files

dengemann, 2017-04-18:
For some CTF files channel names contain run-specific suffixes. This means that for combining data and noise MEG into an inverse solver, users have to know that REF channels must be kept all the time, and all channels must be cleaned, e.g. using `mne.utils._clean_names`, and this not only for the channel names and ref channel names exposed by the raw object but also inside the info, especially the compensation matrices. This could benefit from some abstraction and simplification. I'd suggest at least one function to clean up CTF data if needed.

cc @agramfort @Eric89GXL

dengemann, 2017-04-18:
currently, all over the place, one has to do things like:

```Python

raw.rename_channels(
    dict(zip(raw.ch_names, mne.utils._clean_names(raw.ch_names))))

for comp in raw.info['comps']:
    for key in ('row_names', 'col_names'):
        comp['data'][key] = mne.utils._clean_names(comp['data'][key])

raw_noise.rename_channels(
    dict(zip(raw_noise.ch_names,
             mne.utils._clean_names(raw_noise.ch_names))))
```

jona-sassenhagen, 2017-04-18:
If I understand correctly, you're bringing this up now because I just sent you some very bad code of mine where I'm badly dealing with that problem? :D

dengemann, 2017-04-18:
> If I understand correctly, you're bringing this up now because I just sent you some very bad code of mine where I'm badly dealing with that problem

No it's a pure coincidence. I have been dealing with CTF data recently and was summarising how I got things done.

Eric89GXL, 2017-04-18:
This can be fixed by:

1. `raw.rename_channels` should at the very least be made smart enough to also change `raw.info['comps']`.

2. Adding `raw.clean_names()` method that internally calls `raw.rename_channels` after doing a clean_names operation.

Then for your use case you just do `raw.clean_names()` and you're done.

dengemann, 2017-04-18:
+1 yes, this sounds like a good API!

OlehKSS, 2018-03-09:
Hello! I have created a [pull request](https://github.com/mne-tools/mne-python/pull/5004) to tackle this issue. Could you take a look?

I am new to MNE and just getting around the project, so I have several questions. 
First of all, I have  added clean_names function to the SetChannelsMixin. Should it be in this class or should I move it to BaseRaw (or Raw) class, since the mixin can be used for Evoked and Epochs data types as well?

Should I write unit tests for this method? I was looking around for rename_channels method tests, but I couldn’t find any. There are tests for rename_channels function. Could you also provide me with sample data for the testing?

dengemann, 2018-03-09:
Let's discuss in the PR you created.

larsoner, 2018-04-04:
Fixed by #5004

closed 2018-04-04 by larsoner