mne-tools/mne-python #4208: Creating stimulus channel from scratch / adding events to the raw object

ryscet, 2017-04-19:
I have data from experiment in two different files, first and .edf file with EEG data and second a csv file with events. I would like to create a  mne.Raw (`mne.io.read_raw_edf()`) object and afterwards add a stimulus channel to it. 

From the csv file I created an array of `events` which I would like to  add the the mne.Raw using `raw.add_events()`. This however requires a stimulus channel which is not there, so it needs to be created. I couldn't find an example or documentation describing how to create a stimulus channel from scratch or more generally on adding events from another file. Is events channel an array of 0's plus a value coding for event on all event indexes?

I understand I could create epochs data using the `raw` object created from edf file and `events` from csv file but for convenience I would like to store them in a single mne object, is there such possibility?

jaeilepp, 2017-04-19:
You can use [add_channels](https://mne-tools.github.io/dev/generated/mne.io.Raw.html?highlight=add_channels#mne.io.Raw.add_channels)

So the flow would be something like this:
```
info = mne.create_info(['STI'], raw.info['sfreq'], ['stim'])
stim_raw = mne.io.RawArray(stim_data, info)
raw.add_channels([stim_raw], force_update_info=True)
```

ryscet, 2017-04-19:
Okay, thanks! How should stim_data look like?

jaeilepp, 2017-04-19:
It should be 2d array. So something like
```
stim_data = np.zeros((1, len(raw.times))
```

ryscet, 2017-04-19:
Great, that worked. I didn't know what was the format of the stim_data, now it's clear. Perhaps this should go into documentation, I could make an example of how to add events from a separate file and a submit a pull request.

closed 2017-04-19 by ryscet

Eric89GXL, 2017-04-19:
That would be helpful @ryscet

piotrdzwiniel, 2018-03-01:
It would be great to include in the documentation information about the required shape of raw objects send to `add_channels` function in `add_list` list parameter. When adding only one channel it is not clear to represent this one-dimensional channel as two-dimensional matrix `n_channels, n_samples` as for example `(1, n_samples)`. It's not intuitive.

girijesh97, 2022-02-24:
Hi @larsoner @ryscet @jaeilepp, I am working on a EEG data which is available here (https://git.scc.kit.edu/pd7949/brainwave-authentication). I am also stuck in the similar problem asked by @ryscet. I have three different files for each individual that contains EEG data, Events and time stamp. Likewise, I tried the method that has been mentioned here and here [https://mne.tools/stable/auto_tutorials/raw/20_event_arrays.html]. But I am not able to understand how to read the events, and then merge with the raw data for further analysis. Sorry if this question appears to be naive because I am pretty new to brain signals, so any suggestion or help would be highly appreciated.

larsoner, 2022-02-24:
@girijesh97 this seems more like a question on usage, can you post it to https://mne.discourse.group ?