mne-tools/mne-python #4214: `set_bipolar_reference` fails for a specific montage

jbschiratti, 2017-04-20:
In the following example, I would like to use `mne.set_bipolar_reference` to define a bipolar montage which is sometimes called "longitudinal bipolar anterior-posterior bipolar montage". From the list of electrodes (see fig below), new channels are to be created as follow : 
F3 - C3
C3 - P3
P3 - O1
F4 - C4
C4 - P4
P4 - O2
F3 - FC5
FC5 - T7
... and so on. These new channels would be created by the function `set_bipolar_reference` by providing the right list of channels as `anode` and `cathode`. However, in this montage, F3 appears two times in `anode`. In `set_bipolar_reference`, when `an` is F3, the channel F3 is replaced with F3 - C3. Later, when we need F3 to create F3 - FC5, the channel F3 is gone and the function raises an error.

[image]

    import os.path as op
    from mne.datasets import testing
    from mne import set_bipolar_reference
    from mne.io import read_raw_eeglab
    base_dir = op.join(testing.data_path(download=False), 'EEGLAB')
    raw_fname = op.join(base_dir, 'test_raw.set')
    montage = op.join(base_dir, 'test_chans.locs')

    event_id = {'rt': 1, 'square': 2}
    raw = read_raw_eeglab(input_fname=raw_fname, montage=montage, 
        event_id=event_id, preload=True)
    anode = ['F3', 'C3', 'P3', 'F4', 'C4', 'P4', 'F3', 'FC5', 'T7', 'P7',
             'PO7', 'F4', 'FC6', 'T8', 'P8', 'PO8']
    cathode = ['C3', 'P3', 'O1', 'C4', 'P4', 'O2', 'FC5', 'T7', 'P7', 'PO7',
               'O1', 'FC6', 'T8', 'P8', 'PO8', 'O2']
    raw = set_bipolar_reference(raw, anode, cathode)

If you want, I can suggest a PR to solve this issue.

Eric89GXL, 2017-04-20:
@wmvanvliet I think you worked on this a bit, can you comment?

wmvanvliet, 2017-04-20:
Looks like we need a `replace_channels=False` option. PR welcome!

jona-sassenhagen, 2017-04-21:
We need this for some sEEG montages too ...

agramfort, 2017-04-22:
@jbschiratti can you open a WIP PR so you can get help ?

agramfort, 2017-07-21:
@jbschiratti can I close this?

jbschiratti, 2017-07-22:
Sure! Sorry, I should have closed it before.

closed 2017-07-22 by agramfort