mne-tools/mne-python #4215: Inverting rank deficient prewhitener

jdue, 2017-04-22:
I have experienced some problems using the ica.apply() method when fitting the ica on average referenced eeg data. This data is rank deficient compared to the original data and has rank n_chan-1 (whether an explicit reference is added or not). This causes the noise covariance matrix to also have rank n_chan-1 and the prewhitener as well. However, scipy.linalg.pinv does not seems to recognize this loss of rank resulting in a good solution to the system AX=I but a rather numerically unstable inverse. Hence, using ica.apply(raw) [keeping all components] does not simply return the original data (as I would have expected) whereas using pinv2 does. Might it be an idea to use this instead (or perhaps pinvh)?

Please note that I am no expert in linear algebra so feel free to ignore this if irrelevant (or supplying average referenced data to ica is wrong in the first place).

[image]

dengemann, 2017-04-22:
That's funny ... I am convinced I computed ICA on EEG with average
referencing dozens of times. Your colormaps suggest that your python
distribution is recent. Can you demonstrate this effect with a snippet
using the MNE sample data?

jdue, 2017-04-22:
Okay, that's strange. The following code (modified from http://martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ica.html) replicates the issue on my system at least (tested with python 2.7 and 3.6). MNE version is 0.14. Scipy version is 0.19.0.

```
import numpy as np
import mne
from mne.datasets import sample
from mne.preprocessing import ICA

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, preload=True)
raw.filter(1, 40, n_jobs=2)

raw.pick_types(meg=False, eeg=True) # modified

raw.info["projs"] = [raw.info["projs"][-1]] # added
raw.apply_proj() # added

n_components = 59-1 # modified for stable ica solution
method = 'fastica'
random_state = 23

noise_cov = mne.compute_raw_covariance(raw) # added

ica = ICA(n_components=n_components, method=method, random_state=random_state,
          noise_cov=noise_cov) # modified
ica.fit(raw)

raw2 = raw.copy() # added
ica.apply(raw2) # added
print(np.allclose(raw[:][0], raw2[:][0])) # added
```

jdue, 2017-04-22:
Changing the line `data = fast_dot(linalg.pinv(self._pre_whitener), data)` to `data = fast_dot(linalg.pinv2(self._pre_whitener), data)` or `data = fast_dot(linalg.pinvh(self._pre_whitener), data)` the last line prints True when I run it.

dengemann, 2017-04-22:
Ah. Did you try the default (spherical whitener without covariance)? It
should be satisfying and leads to good results in my experience. The
covariance option is from very early days and not too well tested recently.
It could probably need some care and we might need to regularise.

jdue, 2017-04-22:
Yes, without noise covariance it gives the expected result (I guess in that case there is no inverse involved). If that should work as well I will just use that. Thank you

agramfort, 2017-04-22:
pinv as an rcond parameter that we could adjust higher than the default.

can you try setting it to 1e-9 or something not too small?

jdue, 2017-04-23:
That does the trick---at least in the two cases I tested, given the relative differences in singular values, e.g., 
```
np.linalg.svd(ica._pre_whitener,compute_uv=False) / np.linalg.svd(ica._pre_whitener,compute_uv=False)[0]
array([  1.00000000e+00,   9.76304156e-01,   9.55330802e-01, ...,
         2.81062987e-02,   1.29452602e-02,   2.55106458e-17])
```
for the MNE example referenced above.

agramfort, 2017-04-23:
cool

can you send a PR that adds a good default for rcond param?

jdue, 2017-04-23:
Sure, check out #4219.

larsoner, 2019-08-20:
Fixed by #4219

closed 2019-08-20 by larsoner