mne-tools/mne-python #4217: check that `stim_channel` exists in edf.py

nbara, 2017-04-22:
This is a follow-up to an issue raised in https://github.com/mne-tools/mne-python/pull/4205.

In `read_raw_edf()`, it is possible to specify `stim_channel` as as `str`. However, if stim_channel doesn't exactly match one of ch_names (e.g. if you input 'status' instead of 'Status'), the code fails [line 517](https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L518) because stim_channel is a `str`, instead of an index.

Technically stim_channel should get converted into an index at [line 463](https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L463), but this never happens if the labels don't match exactly.

I suggest the following addition to `_get_edf_info()` to check that `stim_channel` is actually found in the data, and output a more helpful error message. 

```
from difflib import get_close_matches

if isinstance(stim_channel, str):
    if stim_channel not in ch_names:
        matches = get_close_matches(stim_channel, ch_names, n=1)[0]
        raise ValueError('Could not find a channel named "%s" in datafile'
                         ', closest match is "%s"' % (stim_channel, matches))
```

If you think this is a good idea, I can make a PR once #4205 is merged, or squeeze this into #4205.

agramfort, 2017-04-22:
fine with me but I think that get_close_matches is a too loose check. I
would just check that strings match in lower/upper case eventually omitting
spaces.

nbara, 2017-04-24:
Ok, so what about this then?

```
    if isinstance(stim_channel, str):
        if stim_channel not in ch_names:
            err = 'Could not find a channel named "%s" in datafile.' \
                  % stim_channel
            casematch = [ch for ch in ch_names
                         if stim_channel.lower().replace(' ', '') == \
                         ch.lower().replace(' ', '')]
            if casematch:
                    err += ' Closest match is "%s".' % casematch[0]
            raise ValueError(err)
```

Is it ok to include this in the open PR?

agramfort, 2017-04-24:
ok

I would make a tiny private subfunction to do this though

nbara, 2017-04-29:
Closed in favour of #4205

closed 2017-04-29 by nbara