mne-tools/mne-python #4222: Cannot install dev MNE (macOS + no Anaconda)

gacek91, 2017-04-24:
Hey there,
I'm trying to install dev version of MNE and I just can't get it to work. I'm getting:   `No module named setuptools` 
Needless to say that not only do I have setuptools installed, but no problems happen when I try to install stable MNE. Two possibly significant factors:
1. macOS user
2. not using Anaconda - I couldn't find any other way that would enable keyboard shortcuts in `raw.plot()`

Any ideas?

Regards,

Maciej Gaca

agramfort, 2017-04-24:
how did you install python with scipy etc.?

what does

which python
which pip

give you ?

gacek91, 2017-04-24:
Used `pip3 install <package>`.

`which python3`:  `/Library/Frameworks/Python.framework/Versions/3.6/bin/python3`
`which pip3 `:  `/Library/Frameworks/Python.framework/Versions/3.6/bin/pip3`

`which python` and `which pip` are connected to Python 2.7 so I suppose they aren't of any use atm.

agramfort, 2017-04-24:
so are you installing with

python3 setup.py install

?

the dev version too?

gacek91, 2017-04-24:
Of course I did **not**. Well that was easy.

Thank you very much, apologies for that.

Regards

closed 2017-04-24 by gacek91