mne-tools/mne-python #4228: API 'QDate' has already been set to version 1

er06645810, 2017-04-27:
I'm running into a crash using `mne.vis.plot` in one of the tutorials. I believe it is a dependency issue, but it's strange because I'm getting the same crash on both Windows and Fedora.

On both I created a new virtual env in Anaconda based on Python 2.7 and installed
`conda install scipy matplotlib scikit-learn mayavi ipython-notebook`
`pip install PySurfer`

On the Fedora install, I also had to run
`conda install nomkl numpy scipy scikit-learn numexpr`
`conda remove mkl mkl-service` 
to resolve issues with MKL. 

The Windows install just uses the latest stable version of mne from pip, and the Fedora install is from the github source.

Stack Trace
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-9d098eb31f56> in <module>()
      3 
      4 info = mne.io.read_info(raw_fname)
----> 5 mne.viz.plot_trans(info, trans, subject, dig=True, subjects_dir=subjects_dir)

/home/erik/mne-python/mne/viz/_3d.pyc in plot_trans(info, trans, subject, subjects_dir, source, coord_frame, meg_sensors, eeg_sensors, dig, ref_meg, ecog_sensors, head, brain, skull, src, mri_fiducials, verbose)

/home/erik/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

/home/erik/mne-python/mne/viz/_3d.pyc in plot_trans(info, trans, subject, subjects_dir, source, coord_frame, meg_sensors, eeg_sensors, dig, ref_meg, ecog_sensors, head, brain, skull, src, mri_fiducials, verbose)
    472     """
    473     from ..forward import _create_meg_coils
--> 474     mlab = _import_mlab()
    475     if meg_sensors is False:  # old behavior
    476         meg_sensors = 'helmet'

/home/erik/mne-python/mne/utils.pyc in _import_mlab()
   1149     """Quietly import mlab."""
   1150     with warnings.catch_warnings(record=True):
-> 1151         from mayavi import mlab
   1152     return mlab
   1153 

/home/erik/anaconda2/lib/python2.7/site-packages/mayavi/mlab.py in <module>()
     25 
     26 # Mayavi imports
---> 27 from mayavi.tools.camera import view, roll, yaw, pitch, move
     28 from mayavi.tools.figure import figure, clf, gcf, savefig, \
     29     draw, sync_camera, close, screenshot

/home/erik/anaconda2/lib/python2.7/site-packages/mayavi/tools/camera.py in <module>()
     23 # We can't use gcf, as it creates a circular import in camera management
     24 # routines.
---> 25 from .engine_manager import get_engine
     26 
     27 

/home/erik/anaconda2/lib/python2.7/site-packages/mayavi/tools/engine_manager.py in <module>()
     10 
     11 # Local imports
---> 12 from mayavi.preferences.api import preference_manager
     13 from mayavi.core.registry import registry
     14 from mayavi.core.engine import Engine

/home/erik/anaconda2/lib/python2.7/site-packages/mayavi/preferences/api.py in <module>()
      2 
      3 # The global PreferenceManager instance
----> 4 from .preference_manager import preference_manager
      5 from .bindings import set_scene_preferences, get_scene_preferences

/home/erik/anaconda2/lib/python2.7/site-packages/mayavi/preferences/preference_manager.py in <module>()
     27 from traits.etsconfig.api import ETSConfig
     28 from traits.api import HasTraits, Instance
---> 29 from traitsui.api import View, Group, Item
     30 from apptools.preferences.api import (ScopedPreferences, IPreferences,
     31         PreferencesHelper)

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/api.py in <module>()
     34 
     35 try:
---> 36     from .editors.api import ArrayEditor
     37 except ImportError:
     38     # ArrayEditor depends on numpy, so ignore if numpy is not present.

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/editors/__init__.py in <module>()
     21 
     22 try:
---> 23     from .api import ArrayEditor
     24 except ImportError:
     25     pass

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/editors/api.py in <module>()
     22 from .button_editor import ButtonEditor
     23 from .check_list_editor import CheckListEditor
---> 24 from .code_editor import CodeEditor
     25 from .color_editor import ColorEditor
     26 from .compound_editor import CompoundEditor

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/editors/code_editor.py in <module>()
     34 #-------------------------------------------------------------------------------
     35 
---> 36 class ToolkitEditorFactory ( EditorFactory ):
     37     """ Editor factory for code editors.
     38     """

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/editors/code_editor.py in ToolkitEditorFactory()
     46 
     47     # Background color for marking lines
---> 48     mark_color = Color( 0xECE9D8 )
     49 
     50     # Object trait containing the currently selected line (optional)

/home/erik/anaconda2/lib/python2.7/site-packages/traits/traits.pyc in __call__(self, *args, **metadata)
    520 
    521     def __call__ ( self, *args, **metadata ):
--> 522         return self.maker_function( *args, **metadata )
    523 
    524 class TraitImportError ( TraitFactory ):

/home/erik/anaconda2/lib/python2.7/site-packages/traits/traits.pyc in Color(*args, **metadata)
   1234     from traitsui.toolkit_traits import ColorTrait
   1235 
-> 1236     return ColorTrait( *args, **metadata )
   1237 
   1238 Color = TraitFactory( Color )

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/toolkit_traits.pyc in ColorTrait(*args, **traits)
      5 
      6 def ColorTrait ( *args, **traits ):
----> 7     return toolkit().color_trait( *args, **traits )
      8 
      9 def RGBColorTrait ( *args, **traits ):

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/toolkit.pyc in toolkit(*toolkits)
    159             try:
    160                 with provisional_toolkit(toolkit_name):
--> 161                     _toolkit = _import_toolkit(toolkit_name)
    162                     return _toolkit
    163             except (AttributeError, ImportError) as exc:

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/toolkit.pyc in _import_toolkit(name)
     81 
     82 def _import_toolkit ( name ):
---> 83     return __import__( name, globals=globals(), level=1 ).toolkit
     84 
     85 

/home/erik/anaconda2/lib/python2.7/site-packages/traitsui/qt4/__init__.py in <module>()
     16 # import pyface.qt before anything else is done so the sipapi
     17 # can be set correctly if needed
---> 18 import pyface.qt
     19 
     20 #----------------------------------------------------------------------------

/home/erik/anaconda2/lib/python2.7/site-packages/pyface/qt/__init__.py in <module>()
     31     except ImportError:
     32         try:
---> 33             prepare_pyqt4()
     34             import PyQt4
     35             qt_api = 'pyqt'

/home/erik/anaconda2/lib/python2.7/site-packages/pyface/qt/__init__.py in prepare_pyqt4()
     15     # Set PySide compatible APIs.
     16     import sip
---> 17     sip.setapi('QDate', 2)
     18     sip.setapi('QDateTime', 2)
     19     sip.setapi('QString', 2)

ValueError: API 'QDate' has already been set to version 1
```

jaeilepp, 2017-04-27:
I believe this is because of a conflict with mayavi. I get around it by calling ``from pyface.qt import QtGui, QtCore`` before importing mayavi.

er06645810, 2017-04-27:
That got me a little bit further! A window appears for a split second before the kernel crashes with no error.

jaeilepp, 2017-04-27:
I would start debugging by seeing if mayavi works like it should. Can you try running the example [here](http://docs.enthought.com/mayavi/mayavi/auto/mlab_helper_functions.html#mayavi.mlab.plot3d).

er06645810, 2017-04-27:
Good intuition! Looks like something is wrong with mayavi. I get the same crash with that example function.

er06645810, 2017-04-27:
I'm going to close this issue because it's not related to mne. I will post back again if I find a solution though, incase anybody runs into the same problem again.

closed 2017-04-27 by er06645810

Eric89GXL, 2017-04-27:
You can also run with a different ETS_TOOLKIT like wx