mne-tools/mne-python #4229: BUG: coregistration translation / rotation not working

kingjr, 2017-04-27:
The rotation and translation changes seem to have no consequence. I'm not sure whether this is because these options don't update the points, or whether it's a pur viz issue.

- Checking out what changed in the previous PRs, #3940 seems to have fixed an analogous failure to update (viz of?) the *scaling*.

- I use the default fsaverage wiht the following hand-made electrode locations: https://ufile.io/2ctfn

- Adding fiducials didn't seem to change the problem.

- there is no warning or error displayed in the terminal

[image]

@christianbrodbeck

kingjr, 2017-04-27:
The problem disappears when I add nasion/lpa/rpa in the digmontage => https://ufile.io/4i4x9

It seems that the translation and rotation are dependent on these variable (but IMO, they shouldn't).

christianbrodbeck, 2017-04-27:
Yes, we should probably show an error message when a digitizer instance without nasion is specified; coregistration with a center other than nasion would have to be implemented (see https://github.com/mne-tools/mne-python/issues/3934#issuecomment-275425957)

kingjr, 2017-05-11:
I took me a while to understand why I could only work the fix for 'sample' but not 'fsaverage'. Turns out, I need to create a mock `subjects/fsaverage/bem/fsaverage-fiducials.fif` file.

IMO we shouldn't throw an error if nasion etc are missing. You should be allowed to do coregistration purely on the sensors if you know where they are.

christianbrodbeck, 2017-05-12:
> IMO we shouldn't throw an error if nasion etc are missing. You should be allowed to do coregistration purely on the sensors if you know where they are.

Agreed, I suggested a temporary solution until someone can work on it

larsoner, 2018-04-04:
I think this has been fixed by recent coreg changes

closed 2018-04-04 by larsoner