mne-tools/mne-python #4231: montage topoplot differs from eeg_projection plot

pbrusco, 2017-04-27:
Hi! 

I'm having trouble plotting topoplots. 

Executing:
```python
montage_file = "/tmp/LNI.sfp"
montage = mne.channels.read_montage(montage_file)
```

I get 
[image]

which seems to be perfect. 

But then, when I execute 
```python
info = mne.create_info(montage.ch_names, sfreq=512.0, ch_types="eeg", montage=montage)
eeg_layout = mne.channels.make_eeg_layout(info)
eeg_layout.plot()
```
I get:

[image]

which is wrong (90 degrees wrong (?) )

Any ideas?

I have the same problem with plot_topomap that uses the same 2d projection. 

Here the montage_file: https://gist.github.com/pbrusco/b55824af996652156e6ecfcfc830de05

jaeilepp, 2017-04-28:
Looks like your montage file is not in RAS coordinates. You need to switch x and y coordinates (and flip left to right by the looks of it). I wonder how common this is. Maybe there should be an option for switching between coordinate systems.

Eric89GXL, 2017-04-28:
It could be a coordinate system problem, or it could just be that our reader assumes something about the ordering of data in the file format, and it's violated here. @pbrusco do you have time to dig into our code a bit to find out?

pbrusco, 2017-04-28:
Hi! Thanks for the responses. 

I'm not sure what should I check. 

I'm using a .sfp file (that is a EEGlab supported format (matlab)).

When I plot it using Matlab I see the following. 

[image]

The file is exactly what I posted on: https://gist.github.com/pbrusco/b55824af996652156e6ecfcfc830de05

Thanks! let me know how can I help specifically if you need.

agramfort, 2017-04-29:
see this:

https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L91

you need indeed to do something equivalent to:

locs = np.r_[-loc_y, loc_x, loc_z]

as EEGLAB uses a different coordinate system by default

pbrusco, 2017-05-02:
Hi, thanks for the information. 

But I have a doubt anyway. Why is it that montage.plot(show_names=True) show the right positions of the electrodes. Isn't that inconsistent? 

Changing montage.pos = new_positions works. 

Thanks a lot!

closed 2017-05-02 by pbrusco

agramfort, 2017-05-03:
are all the sfp you have access to flipped the same way?

i think we need a fix or a documentation

pbrusco, 2017-05-04:
I'm not sure actually. That's the standard for EEGLab sfp I think but I'm not sure in general.

agramfort, 2017-05-04:
can you share a few of such files so we can test and avoid this issue in
the future?