mne-tools/mne-python #4232: epochs_image: smoothing happens before sorting, extreme overlay_times handling

jona-sassenhagen, 2017-04-28:
1. I think epochs.plot_image could handle `overlay_times` exceeding the data window better:

[image]

E.g., either the x axis of the ERP below should be scaled to the image, or the extreme overlay_times should be dropped, or something like that.

2. I think the image interpolation/smoothing should happen on the *sorted* image, cause otherwise the sorting essentially doesn't do anything:

[image]

@agramfort do you agree?

agramfort, 2017-04-29:
agreed for both. For 2 it's a bug. Smoothing has to be done after sorting.

jona-sassenhagen, 2017-04-29:
Fix for # 2 in #4227. For # 1, what should we do - censor the overlay_times? Drop the epochs? Keep the x axis for the image and the evoked plot aligned?

jona-sassenhagen, 2017-09-06:
Closed by #4227 and #4535

closed 2017-09-06 by jona-sassenhagen