mne-tools/mne-python #4234: better automatic layout for evoked.plot

agramfort, 2017-04-29:
try to run

```
import os.path as op
import numpy as np
import matplotlib.pyplot as plt

import mne

data_path = mne.datasets.sample.data_path()
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True, condition=0)

fig = evoked.plot(selectable=False)
fig = evoked.copy().pick_types(meg=True, eeg=True).plot(selectable=False)
fig = evoked.copy().pick_types(meg="mag", eeg=True).plot(selectable=False)
fig = evoked.copy().pick_types(meg="grad", eeg=True).plot(selectable=False)
fig = evoked.copy().pick_types(meg="grad").plot(selectable=False)
fig = evoked.copy().pick_types(meg=False, eeg=True).plot(selectable=False)
fig = evoked.copy().pick_types(meg=False, eeg=True, eog=True).plot(selectable=False)
```

and you'll see some texts overlapping, some white spaces etc... So it does not look nice automatically.

I can hack it with:

```
import os.path as op
import numpy as np
import matplotlib.pyplot as plt

import mne

data_path = mne.datasets.sample.data_path()
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True, condition=0)

# fig = evoked.plot(selectable=False)
# fig = evoked.copy().pick_types(meg=True, eeg=True).plot(selectable=False)
# fig = evoked.copy().pick_types(meg="mag", eeg=True).plot(selectable=False)
# fig = evoked.copy().pick_types(meg="grad", eeg=True).plot(selectable=False)
# fig = evoked.copy().pick_types(meg="grad").plot(selectable=False)
# fig = evoked.copy().pick_types(meg=False, eeg=True).plot(selectable=False)
# fig = evoked.copy().pick_types(meg=False, eeg=True, eog=True).plot(selectable=False)

fig = evoked.plot(selectable=False)
fig.set_size_inches(6.4, 6)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg=True, eeg=True).plot(selectable=False)
fig.set_size_inches(6.4, 4.5)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg="mag", eeg=True).plot(selectable=False)
fig.set_size_inches(6.4, 3.4)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg="grad", eeg=True).plot(selectable=False)
fig.set_size_inches(6.4, 3.4)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg="grad").plot(selectable=False)
fig.set_size_inches(6.4, 2)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg=False, eeg=True).plot(selectable=False)
fig.set_size_inches(6.4, 2)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
fig = evoked.copy().pick_types(meg=False, eeg=True, eog=True).plot(selectable=False)
fig.set_size_inches(6.4, 3.4)
for ax in fig.axes[:-1]: ax.set_xlabel(' ')
fig.tight_layout()
```

we should make this look nice automatically.

dengemann, 2017-04-29:
in my experience tight_layout is something that is prone to break across
systems and version numbers ...
If you want less user complaints teach them how to clean it up and increase
matplotlib literacy instead of hiding these hacks. But maybe if it's one
shot that does not make things worse if it fails, why not.
My 2 cents.

agramfort, 2017-04-29:
then set the layout params explicitly. I was doing this in the early days
of MNE-Python as it was indeed more robust than tight_layout

dengemann, 2017-04-29:
Let's see what are the results of this poll. Maybe things improved in the
meantime.

Eric89GXL, 2017-04-30:
I usually see decent results with `tight_layout` and we use it in the vast majority of our plotting functions. So +1 for using it / improving our use of it where possible.

agramfort, 2017-05-01:
good

@jaeilepp can you maybe take care of it?

closed 2017-05-04 by agramfort