mne-tools/mne-python #4240: add plot_sphere function similar to plot_bem

agramfort, 2017-05-03:
we need a function to visualize the sphere volume conductor model

something like this:

```
import os.path as op
import mne

data_path = mne.datasets.sample.data_path()
fname = op.join(data_path, 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(fname, baseline=(None, 0), proj=True, condition=0)
evoked.pick_types(meg=False, eeg=True, eog=True)
evoked.set_eeg_reference()

evoked.plot()
evoked.plot_topomap(times=[0.09])

subjects_dir = data_path + '/subjects'
subject = 'sample'
trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'

print('make sphere model')
sphere = mne.make_sphere_model(info=evoked.info, r0='auto',
                               head_radius='auto',
                               relative_radii=[0.8353, 0.847, 0.93, 1],
                               sigmas=(0.33, 1.0, 0.0042, 0.33))
print("[done]")

info = evoked.info
mne.viz.plot_trans(info, trans, subject=subject, dig=True,
                   meg_sensors=False, subjects_dir=subjects_dir,
                   coord_frame='head', head=False,
                   skull=['inner_skull', 'outer_skull'],
                   brain=True)

from mayavi import mlab
center = sphere['r0']
# radius = sphere['layers'][-1]['rad']
mlab.points3d(*center, scale_factor=0.15)
```

it gives:

[image]

but here the radius is **not** the correct one.

Eric89GXL, 2017-05-03:
We can just make `plot_bem` do this. Instead of passing a BEM you pass the sphere model. This is similar to how it works when doing a forward model.

Eric89GXL, 2017-05-03:
... or plot_trans, or whatever we use for BEMs

Eric89GXL, 2017-05-31:
Actually we should just change `plot_trans` to allow `skull=SphereModel` then we're all set, right?

agramfort, 2017-05-31:
I am concerned about API simplicity

setting skull to a sphere BEM seems weird

Eric89GXL, 2017-05-31:
Our `plot_trans` API is not so good. Really we should have a `BEM` parameter or so, and be able to pass a BEM model (list of dict), solution (read from FIF or generated), or SphereModel. We could think about deprecating `skull`, but the BEM/sphere interacts with `head` in the case of a 3-layer BEM or multi-layer SphereModel.

agramfort, 2017-05-31:
+1 for adding a bem parameter and maybe preventing the setting of skull or
head when bem is passed? kind of exclusive rule?

closed 2017-08-03 by agramfort