mne-tools/mne-python #4241: Reading CTF Evokeds fails: duplicate compensation channel

jona-sassenhagen, 2017-05-03:
On master, the attached file was created with `evoked.save(fname)`, but can't be read with `mne.read_evokeds(fname)`:

```
/Users/jona/tools/mne-python/mne/io/meas_info.py in read_meas_info(fid, tree, clean_bads, verbose)
    775
    776     #   Load the CTF compensation data
--> 777     comps = read_ctf_comp(fid, meas_info, chs)
    778
    779     #   Load the bad channel list

/Users/jona/tools/mne-python/mne/io/ctf_comp.py in read_ctf_comp(fid, node, chs, verbose)

/Users/jona/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708
    709

/Users/jona/tools/mne-python/mne/io/ctf_comp.py in read_ctf_comp(fid, node, chs, verbose)
    109         if not calibrated:
    110             #   Calibrate...
--> 111             _calibrate_comp(one, chs, mat['row_names'], mat['col_names'])
    112         else:
    113             one['rowcals'] = np.ones(mat['data'].shape[0], dtype=np.float)

/Users/jona/tools/mne-python/mne/io/ctf_comp.py in _calibrate_comp(comp, chs, row_names, col_names, mult_keys, flip)
     42             if p != 1:
     43                 raise RuntimeError('Channel %s does not appear exactly once '
---> 44                                    'in data' % names[ii])
     45             idx = ch_names.index(names[ii])
     46             val = chs[idx][mult_keys[0]] * chs[idx][mult_keys[1]]

RuntimeError: Channel BG1-3609 does not appear exactly once in data
```

The instance was created by `linear_regression_raw` out of a Raw instance.

[HA0710_demo-ave.fif.zip](https://github.com/mne-tools/mne-python/files/974780/HA0710_demo-ave.fif.zip)

@jaeilepp ..?

Eric89GXL, 2017-05-04:
Can you see if you can reproduce using one of the CTF datasets in `testing`?

If not, can you at least see if this happens without using `linear_regression_raw` (i.e., using plain `Epochs` and `epochs.average()`? My guess is that there is some interaction with `picks` where `linear_regression_raw` chooses non-reference channels, and the resulting `Evoked` instance doesn't have the channels necessary for `comps`, and this is only noticed on read.

jona-sassenhagen, 2017-05-04:
It only depends on picking `ref_meg = False`. E.g., if I write the raw file's info to the disk after picking, I can't read it due to this error. If I don't pick, it works.

jaeilepp, 2017-05-04:
I'm not sure what's the right fix for this. I see 3 options:

1. Don't save the comps from dropped channels.
2. Save the comps, but ignore them while reading, throwing a warning.
3. Not allow saving when required channels are missing.

jaeilepp, 2017-05-04:
And BTW the issue heading is incorrect. The problem is that the compensation channel is not there.

jona-sassenhagen, 2017-05-04:
At a minimum, if the error message read "missing compensation channel" instead of saying that a specific weirdly named channel appears more or less than 1 times, it would be a much easier problem :)
And a warning when removing the comp channels about what consequences this will have - that it precludes file saving/reading, and if it worsens the data later on. (If I understand it, the compensation has already been appplied and they can in principle be safely dropped?)

Eric89GXL, 2017-05-04:
Comps are also necessary for fwd/inv.

My proposed fix is that when picking is done (e.g., in `raw.pick_types`, or in `linear_regresison_raw`, etc.):

1. If compensation has been applied but not all used comp channels are picked, raise an error.

2. If no compensation has been applied, discard the projectors (alongside a `logger.info` that this is being done).

jona-sassenhagen, 2017-05-04:
You have to manually turn off compensation right? If yes - seems sensible to me; if you've manually turned off comps and pick the channels, you probably know what you're doing. Otherwise I would prefer a warning.

dokato, 2017-12-13:
Hi, is there a quick workaround before it gets fixed? I got the same error when I try to read my epoched data, so I assume it's related.

jona-sassenhagen, 2017-12-13:
If you can, it would probably be easiest to recreate the epochs in a what that can be read.

agramfort, 2017-12-13:
yes the easy fix is to avoid storing the data with picked channels.

PR #5083 (2018-04-03): MRG: Fix CTF picking
Implements the fix discussed in https://github.com/mne-tools/mne-python/issues/4241#issuecomment-299203201.

Closes #4241.

@jona-sassenhagen can you see if it makes sense for your data?
closed 2018-04-03 by agramfort