mne-tools/mne-python #4251: Impossible to parallelize compute_covariance

cmoutard, 2017-05-09:
While trying to parallelize mne.cov.compute_covariance (i.e. n_jobs>1), I get the following error in the frame of the function retrieve within sklearn's parallel.py file:

PicklingError: Can't pickle <class 'mne.cov._ShrunkCovariance'>: it's not found as mne.cov._ShrunkCovariance

It looks like this is due to the fact that class _ShrunkCovariance is nested in a function.
Nested class/function is known to create issue with pickle.
If this is not my mistake as a user, I see two options:
1) to unnest this class (and possibly _RegCovariance ?)
2) to follow the suggestion of Pelson [here](http://stackoverflow.com/questions/1947904/how-can-i-pickle-a-nested-class-in-python)

dengemann, 2017-05-09:
Thanks for reporting. For now use 1 job :) I need to find a moment to fix
this. Out if curiosity which Python version do you use?

cmoutard, 2017-05-09:
Issue replicated with Python 2.7.12 and 2.7.13

dengemann, 2017-05-09:
This is really weird. I guess there is a problem with the class being
returned by a function in the cov.py code. Probably it should just be
written as a private class at the module level.

Eric89GXL, 2017-05-09:
We can't do option 1 without importing `sklearn` on module import, which is something we've tried to avoid

dengemann, 2017-05-09:
Yes, this was probably the reason why I came up with this funny pattern of
returning a class. It would be interesting to understand when this stopped
working.

agramfort, 2017-05-09:
that's unfortunate... we need to fix

dengemann, 2017-05-11:
Let's see what it takes to do a shrunk class with lazy imports ...
If someone uses the code, he anyways need sklearn. We need to do the
minimum imports to make the class definition work, we don't care if it
breaks on calling/initializing, as this is then a matter of having sklearn
installed.

cmoutard, 2017-05-11:
I have tried to unnest these classes (e.g. adding a base class in __init__ combined with super) and my attempts always met many issues, notably some conflicts between old-style and new-style classes.
Could not find any way to fix it without lazy imports, but I am not a Python guru.

larsoner, 2019-08-21:
This should be fixed in `master`

closed 2019-08-21 by larsoner