mne-tools/mne-python #4252: Importing GDF fails for BCI Competition IV dataset 2b

robintibor, 2017-05-09:
I wanted to import BCI Competition IV dataset  2b and encountered two problems:

1. impedance logic fails
```python
>>> raw = mne.io.read_raw_edf('data/bci-competition-iv/2b/signal/B0101T.gdf', stim_channel=None)
Extracting edf Parameters from data/bci-competition-iv/2b/signal/B0101T.gdf...
GDF file detected
....
/home/schirrmr/code/mne-python/mne/io/edf/edf.py in _read_gdf_header(fname, stim_channel, exclude)
    909             if edf_info['number'] < 2.19:
    910                 impedance = np.fromfile(fid, np.uint8, len(channels))
--> 911                 impedance[impedance == 255] = np.nan
    912                 channel['impedance'] = pow(2, impedance.astype(float) / 8)
    913                 fid.seek(19 * len(channels), 1)  # reserved

ValueError: cannot convert float NaN to integer
```

This I could solve by casting to float, so changing 
```python
impedance = np.fromfile(fid, np.uint8, len(channels))
``` 
to 
```python
impedance = np.fromfile(fid, np.uint8, len(channels)).astype(float)
```
.

2. Events are not being extracted properly?

I don't know if I am correctly trying to extract the event information. According to the dataset description (http://bbci.de/competition/iv/desc_2b.pdf), the header should contain the event information, with 120 trials in one sessionfile.
If I read the data as described above after fixing the impedance problem, and access the raw_extras events field, it only contains 17 events.

```python
>>> raw = mne.io.read_raw_edf('data/bci-competition-iv/2b/signal/B0101T.gdf', stim_channel=None)
>>> raw._raw_extras[0]['events']
[[17, array([     1,    500,  16750,  33000,  38000,  43000,  48000,  52252,
          55140,  55890,  57500,  58250,  62068,  62818,  64105,  64855,
          72982,  73732,  75189,  75939,  77590,  78340,  88501,  88501,
          89251,  90935,  91685,  93194,  93944,  99954, 100704, 107107,
         107857, 109568, 110318, 112029, 112779, 116392, 116392, 117142,
         121162, 121912, 123612, 124362, 132647, 133397, 139432, 139432,
         140182, 141478, 142228, 144344, 144942, 145692, 147232, 147982,
         149592, 150342, 156197, 156947, 160592, 161342, 165074, 165824,
         169682, 170432, 171991, 171991], dtype=uint32), array([41669,     2, 45177,     2, 45927,     2, 47234,     2, 47984,
             2, 51955,     2, 52705,     2, 54214,     2, 54964,     2,
         60974,     2, 61724,     2,  2591,     3,  2591,     3,  3341,
             3, 14308,     3, 15058,     3, 19096,     3], dtype=uint16), array([19846,     3, 23660,     3, 24410,     3, 25850,     3, 26600,
             3, 30586,     3, 31336,     3, 32902,     3, 33652,     3,
         39828,     3, 45076,     3, 45826,     3, 47398,     3, 48148,
             3, 53971,     3, 54721,     3, 56076,     3], dtype=uint16), array([253434, 257166, 257916, 259373, 260123, 264083, 264833, 266135,
         266885, 268341, 269091, 281771, 282521, 284138, 284888, 286577,
         287327, 288989, 289739, 296213, 296963, 298260, 298260, 299010,
         310156, 310906, 314550, 315300, 316831, 317581, 319286, 320036,
         325662, 325662, 326412, 328528, 331416, 331416, 332166, 338344,
         338344, 339094, 342671, 343421, 346803, 347553, 353866, 354616,
         356175, 356925, 358227, 358977, 360433, 361183, 362490, 363240,
         369470, 369470, 370220, 371754, 372504, 373863, 373863, 374613,
         376230, 376980, 378669, 379419], dtype=uint32)]]
```

Am I doing this correctly, if so any ideas why this fails?

Best,
Robin

nbara, 2017-05-11:
Hi,

I wrote the GDF code that's giving you these errors. At the time I wrote the code, I couldn't find any sample GDF files which had event tables so I am not terribly surprised this is where the code breaks.

1. That's definitely a bug. Good catch!
2. That looks very weird indeed. What you can do in the meantime is get the event timings in Matlab, and just import that in python. MNE uses a very straightforward event structure (see http://martinos.org/mne/stable/generated/mne.find_events.html)

The current gdf reader does the same thing as an old [biosig port for python](https://github.com/donnchadh/biosig/blob/master/biosig4python/biosig.py), but maybe that file is outdated as well. 

I don't know if I'll have time to fix this myself right now, but it would help if you could provide a link to the GDF file? Also, whoever fixes this might also want to look at the [gdf reader that ships with Fieldtrip](https://github.com/fieldtrip/fieldtrip/blob/master/external/biosig/sopen.m). AFAIK this is the only working reader out there besides the official C++ biosig code.

robintibor, 2017-05-16:
Hi,
thanks for the answer.
Unfortunately, while it now loads, the events are still not correctly extracted :( The files you can get from http://bbci.de/competition/iv/#download , you have to register but it is automatic, you immediately get an email and a password once you register.

I know I can get the things from matlab, I have done that, but it would be great to do it in python without requiring any matlab :)

closed 2017-06-02 by agramfort