mne-tools/mne-python #4255: DipoleFit head coords to mri coords transformation

d-van-de-velden, 2017-05-10:
Dear all,

I am using your mne-toolbox for a project in my study. We designed an MEG experiment and used a single dipole fit for each hemisphere. the Dipole position is calculated in head coordinates. 
Since we want to compare it to another calculation tool, which only calculates in mri coordinates, we tried to find a tool/function/command that will re-transform our Dipole positions into mri coords.

We couldn't find any.

Did we miss something or could you just tell us what to use ?

We are happy about any hint and/or help.

Yours sincerely,
C. Gurk, D. van de Velden and P. Zwinscher

jona-sassenhagen, 2017-05-10:
@Eric89GXL I will *try* to give an answer today or tomorrow :)

But in short, you probably want to do something like:

- find the affine matrix for the transformation from mri to head coordinates
- invert the matrix
- np.dot(inverse, coordinate)

Take a look at `mne.transforms.Transform`

d-van-de-velden, 2017-05-16:
Hello jona-sassenhagen,

thanks for your advice and i apologize for the long time I needed to reply.
Here is what I got so far:

```
 # Fit a dipoles for 100ms, 150ms and 300ms
dip_100 = mne.fit_dipole(evoked_ave_100, cov, fname_bem, fname_COR)[0]
dip_150 = mne.fit_dipole(evoked_ave_150, cov, fname_bem, fname_COR)[0]
dip_300 = mne.fit_dipole(evoked_ave_300, cov, fname_bem, fname_COR)[0]

 # Find and print Dipole 100ms with best Goodnes of Fit (GOF) and find time, position and orientation
dip_best_100 = np.argmax(dip_100.gof)
dip_time_100 = dip_100.times[dip_best_100]
dip_pos_100 = dip_100.pos[dip_best_100]
dip_ori_100 = dip_100.ori[dip_best_100]

trans = mne.transforms.read_trans(fname_COR)
affine_trans = mne.transforms.invert_transform(trans)
dip_pos100_head = mne.transforms.apply_trans(affine_trans, dip_pos_100)
```

Would you consider the procedure as correct ? 

Thank you for your answer in advance.

Greetings,
C. Gurk, D. van de Velden and P. Zwinscher

agramfort, 2017-07-21:
there is now a plot_dipole_locations

http://www.nmr.mgh.harvard.edu/mne/dev/generated/mne.viz.plot_dipole_locations.html

than can plot with MRI overlays.

closing

closed 2017-07-21 by agramfort