mne-tools/mne-python #4256: Multiscale Sample Entropy (MSE)

KirstenHilger, 2017-05-10:
Dear MNE-Team,

my interest is investigating the association between an ID variable and brain signal variability with EEG. 

Many researches used Multiscale (Sample) Entropy to operational brain signal variability, because it is suggested to measure the predictability of neural operations across the multiple time scales on which the brain operates (for review see Garrett et al., 2013).

Unfortunately, I could not find how to compute MSE with MNE. Would it be possible to implement this entropy measure in MNE or what measure would you recommend to operationalize brain signal variability in general? I would prefer to use an entropy measure (signal complexity) instead of PCA (brain signal variability operationalzed by dimensionality of signal).

MSE was defined in Costa et al. (2005).

Looking forward to your answer!

Best, 
Kirsten

Costa, M., Goldberger, A. L. & Peng, C. K. Multiscale entropy analysis of biological signals. Phys. Rev. E - Stat. Nonlinear, Soft Matter Phys. 71, 1–18 (2005).

Garrett, D. D. et al. Moment-to-moment brain signal variability: A next frontier in human brain mapping? Neurosci. Biobehav. Rev. 37, 610–624 (2013).

DominiqueMakowski, 2017-05-10:
@KirstenHilger, you might want to check [this function](http://neurokit.readthedocs.io/en/latest/documentation.html#complexity) that returns several indices of complexity, including MSE. Let me know how it goes.

kingjr, 2017-05-30:
IMO, there exists many complexity markers in M/EEG, and several may not be as mature as the current signal processing that are implemented in this package.

However, perhaps you can start by making a notebook/git and see whether other dev would find it useful to integrate it as a tutorial. 

cc @dengemann

KirstenHilger, 2017-08-04:
Thank you very much!
The computation with neurokid works fine!

Eric89GXL, 2017-08-04:
@KirstenHilger thanks for reporting back -- feel free to open a PR showing an example of using it if you have time. Closing for now.

closed 2017-08-04 by Eric89GXL