mne-tools/mne-python #4259: Setting title of plot_joint

jona-sassenhagen, 2017-05-11:
Removing the `Sensor_type, X channels` part of the title of a plot_joint fig is surprisingly hard. I also never find it useful ...
E.g., if you do `evoked.plot_joint(ts_args={"titles":"nothing"})`, it will still append `Sensor_type, X channels`. If you access the axis and set its title directly (`fig.axes[0].set_title("x")`), the initial title is unaffected and a new one created! This is specific to `plot_joint`; if you set the title of `evoked.plot`, you set the title.

I'd like to make it easier to remove, or outright remove it.

@jaeilepp

jona-sassenhagen, 2017-05-11:
It's this:
`evoked.plot_joint().axes[2].texts[0].set_text('custom_title')`
which is IMO not straight-forward. There can be a lot of axes in the plot.

One thing I could imagine is attaching the individual plots to the figure with predictable names:
`fig.ts_plot`
`fig.topo_plots` (list of topoplot axes)

WDYT? That would avoid adding a keyword.

Another option is to simply set up `plot_joint` so it behaves like `evoked.plot` wrt. the title, although I don't actually know why it doesn't.

mmagnuski, 2017-05-11:
@jona-sassenhagen I agree, I was using the `texts[0].set_text()` way, but I wasn't happy with it :)

jaeilepp, 2017-05-12:
I think we could treat it as a bug fix and just overwrite the title from now on when ``title is not None``.

closed 2017-08-03 by mmagnuski