mne-tools/mne-python #4261: BUG: raw.plot(show=True) doesn't show

kingjr, 2017-05-11:


jaeilepp, 2017-05-12:
I'm not able to reproduce this. Does ``plt.show`` make it appear?

kingjr, 2017-05-12:
Apparently, it plots the second time I call it, but not the first time:

```python
import mne
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.Raw(fname)
```
```
Opening raw data file /home/jrking/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Current compensation grade : 0
```
```python
raw.plot_sensors(show=True)
```

```
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Out[5]: <matplotlib.figure.Figure at 0x7f922f0bb410>
```

# no plot here

```python
raw.plot_sensors(show=True)
```

# plots

Eric89GXL, 2017-05-31:
Does this behavior change depending on which matplotlib backend you use?

agramfort, 2017-07-22:
I cannot replicate. Closing

closed 2017-07-22 by agramfort