mne-tools/mne-python #4264: bug/suggestion: saving ica objects

jdue, 2017-05-12:
Hi

I ran into an issue when trying to save an ica object to disk. A `PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: [...]` was raised by the save method in mne/preprocessing/ica.py, specifically in the try/except block. However, the reason it entered the except block was that a `TypeError: Object of type 'int32' is not JSON serializable` was raised in _serialize (in _write_ica). Thus, it seems that (at least on Windows on which I am working) if an error occurs while writing the ica object then a `PermissionError` is raised (I guess because the file has not be closed) irrespective of the error causing _write_ica to fail. Furthermore, the reason for the `TypeError` (the 'real' problem in my case) was that I (incidentally) passed a numpy.int32 to ica.n_components.

Thus, I have two suggestions.

First, given that a `PermissionError` was raised instead of the 'actual' error suggest a (perhaps platform specific?) bug which might be fixed, e.g., by changing
```
try:
    _write_ica(fid, self)
except Exception:
    os.remove(fname)
    raise
end_file(fid)
```
to 
```
try:
    _write_ica(fid, self)
    end_file(fid)
except Exception:
    end_file(fid)
    os.remove(fname)
    raise
```
in the ica save method (in ica.py) such that the file is closed before attempting to remove it (this would cause the 'correct' error to be raised by `raise`).

Second, I passed a numpy integer to n_components (particularly, it was a rank estimate as returned by raw.estimate_rank) which resulted in a `TypeError` from json. A possible fix might be to change
```
elif isinstance(value, int):
    value = int(value)
```
to 
```
elif isinstance(value, (int, np.integer)):
    value = int(value)
```
in _serialize (in ica.py) [or perhaps casting to core int when instantiating the ica object?]. I am aware that the doc for mne.preprocessing.ICA actually *do* specify `n_components : int | float | None` but in many applications core and numpy ints behave similarly (correct me if I am wrong) and so one might not be aware whether one is passing one or the other unless explicitly testing it - at least in this case I wasn't :)

The issue can be reproduced using the MNE tutorial from http://martinos.org/mne/stable/auto_tutorials/plot_artifacts_correction_ica.html by replacing the line `n_components = 25` with for example `n_components = np.int32(25)`.

I am using python 3.6 on Windows 10 64-bit.

Eric89GXL, 2017-05-12:
> A possible fix might be to change

We usually use `elif isinstance(value, numbers.Integral)` to be general.

The other fix (closing before raising) seems reasonable to me.

jdue, 2017-05-12:
yes, numbers.Integral is probably better

agramfort, 2017-05-12:
can you send a PR?

jdue, 2017-05-16:
Yes, here it is #4270. I skipped the second part since that was actually fixed about a month ago in #4179 but which I did not have in my local copy. Thus, the PR concerns only the first part of this issue.

closed 2017-07-21 by agramfort