mne-tools/mne-python #4268: BUG: mne.make_field_map indicate wrong `origin` parameter

kingjr, 2017-05-15:
There's a mismatch between docs and actual default parameter. 

I'm not sure what the param should be.

cc @Eric89GXL

Eric89GXL, 2017-05-16:
We usually try to use `'auto'`, so we should probably change it to that as a bugfix

kingjr, 2017-05-16:
does auto take the center of the electrodes or of a sphere fit to the head ?

Eric89GXL, 2017-05-16:
sphere fit to the head digitization

kingjr, 2017-05-16:
I started to make a fix by changing the default parameter value, but it
creates a series of error in the tutorials and in the tests, so I'll let
you guys deal with it

Eric89GXL, 2017-05-16:
For now we can just fix the documentation to reflect the actual default then.

kingjr, 2017-05-16:
(0, 0, 0.04) is a very weird default parameter IMO

Eric89GXL, 2017-05-16:
it is approximately where the origin fit usually ends up for adult subjects

kingjr, 2017-05-16:
yes, I had guessed, but this kind of numerical guess can lead to very
weird-looking results, without a clear warning, so I would recommend doing
either a real fix, or just explicitly use these numbers in the tests and
tutorials that crash with the 'auto' parameter.

commit 0f3416f2 (2017-05-31): FIX: Fix #4268
closed 2017-05-31 by Eric89GXL