mne-tools/mne-python #4275: Add function to plot montage (channel locations)

cbrnr, 2017-05-18:
As mentioned in #4273, it is not straightforward to plot channel locations of a given montage. Here's a function that might be convenient when a user just wants to visualize channel locations without having to load any data:

```Python
import mne
from mne.channels.montage import _set_montage

def plot_montage(montage):
    if isinstance(montage, str):
        montage = mne.channels.read_montage(montage)
    info = mne.create_info(montage.ch_names, sfreq=256, ch_types="eeg")
    _set_montage(info, montage)
    mne.viz.plot_sensors(info, show_names=True)

plot_montage("biosemi32")
```

In which module should this function go? Right now, it only plots EEG channels - are there montages for other channel types available and should these be taken into account as well? Does this function need other arguments (such as an option to show only a subset of channels, an option to hide names, etc)?

jaeilepp, 2017-05-18:
I think it would be appropriate to change the ``plot`` method to reuse the ``plot_sensors`` code. Note that ``create_info`` takes a ``montage`` param you can use instead of calling ``_set_montage``.

cbrnr, 2017-05-18:
Alright, so this becomes:

```Python
import mne

def plot_montage(montage):
    if isinstance(montage, str):
        montage = mne.channels.read_montage(montage)
    info = mne.create_info(montage.ch_names, sfreq=256, ch_types="eeg",
                           montage=montage)
    mne.viz.plot_sensors(info, show_names=True)

plot_montage("biosemi32")
```

Which `plot` method do you mean?

jaeilepp, 2017-05-18:
``montage.plot``

cbrnr, 2017-05-18:
Yes, this would make sense as currently only a 3D plot can be created.

PR #4276 (2017-05-30): [MRG] Use mne.viz.utils.plot_sensors in mne.channels.montage.Montage.plot
Fixes #4275. Refactor `mne.channels.montage.Montage.plot` to use `mne.viz.utils.plot_sensors`. This means that the argument `scale_factor` is currently unused since this is not implemented in that method.
closed 2017-05-30 by Eric89GXL