mne-tools/mne-python #4279: Segmentation fault when computing ICA

cbrnr, 2017-05-22:
tl;dr: randomized PCA from `sklearn` is causing a segmentation error, switching to PCA with `svd_solver='full'` seems to fix this issue.

I'm trying to compute ICA on an EEG dataset with annotations created in the interactive plot window. Everything works fine, but if I append segments between specific events via `raw.annotations.append` I get a segmentation error:

    Python(6414,0x7fffe29793c0) malloc: *** mach_vm_map(size=18446744065369096192) failed (error code=3)
    *** error: can't allocate region
    *** set a breakpoint in malloc_error_break to debug
    Segmentation fault: 11

I've traced the issue as far back as line 258 in `sklearn/utils/extmath`:

    Q, _ = linalg.lu(safe_sparse_dot(A, Q), permute_l=True)

Which is called somewhere along the line by `randomized_svd`, which means that the `PCA` object is created using `svd_solver='randomized'`. I strongly assume that PCA with the randomized solver doesn't work for some reason, because using another solver fixes this problem.

All this is happening with real data during my analysis pipeline, so I don't have a MWE at the moment (and it would probably take me quite a while to create it, so I'm hoping this won't be necessary).

agramfort, 2017-05-22:
can you share the X numpy array that causes the seg fault?

cbrnr, 2017-05-22:
Yes, https://drive.google.com/open?id=0Bzq48JpOZMxLckZKNEpDT0FjXzg

agramfort, 2017-05-23:
it does not crash on my machine

--

In [11]: X = np.load('X_segfault_rand_svd.npz')['X']

In [12]: X.shape
Out[12]: (596626, 62)

In [13]: pca = PCA(svd_solver='randomized')

In [14]: pca.fit(X)
Out[14]:
PCA(copy=True, iterated_power='auto', n_components=None, random_state=None,
  svd_solver='randomized', tol=0.0, whiten=False)

Eric89GXL, 2017-05-23:
Hooray compiled code! `can't allocate region` almost sounds like an out-of-memory error. What kind of hardware do you have? What is your memory and `mne.sys_info()`?

cbrnr, 2017-05-23:
Yes, sounds like there's not enough RAM, but the array is only 283MB and it crashes on my MacBook with 16GB RAM. I just confirmed this behavior on another MacBook (only 8GB RAM), but I get the same crash.

The env is up to date:
```
Platform:      Darwin-16.6.0-x86_64-i386-64bit
Python:        3.6.1 (default, Mar 28 2017, 06:42:56)  [GCC 4.2.1 Compatible Apple LLVM 8.1.0 (clang-802.0.38)]
Executable:    /usr/local/opt/python3/bin/python3.6

mne:           0.14
numpy:         1.12.1 {blas=-msse3, lapack=-msse3}
scipy:         0.19.0
matplotlib:    2.0.2

sklearn:       0.18.1
nibabel:       Not found
mayavi:        Not found
pycuda:        Not found
skcuda:        Not found
pandas:        0.20.1
```

cbrnr, 2017-05-23:
@agramfort I assume you're not on a Mac? I can test this tomorrow on a Linux machine in my office.

agramfort, 2017-05-23:
I am on a mac with anaconda + mkl

Eric89GXL, 2017-05-23:
@cbrnr it looks like you are on some system Python variant. You might have better luck with Anaconda. Plus you'll get MKL, which could potentially give you some nice speedups.

Did you build `sklearn` yourself, then?

cbrnr, 2017-05-23:
No, I've installed Python via Homebrew and all packages via pip. Anaconda gave me some headaches so I'd like to stay away from it for now.

Eric89GXL, 2017-05-23:
> Anaconda gave me some headaches so I'd like to stay away from it for now.

Now you have traded for different headaches :)

But more seriously, did you follow the instructions about installing using wheels?

http://scikit-learn.org/stable/install.html

cbrnr, 2017-05-23:
True :smile:! I'm pretty sure that the binary wheel got installed when I typed `pip3 install scikit-learn` (but I'm aware that I need to perform a manual configuration step when installing from source). This could be related to my BLAS version, which is OpenBLAS and not MKL (or at least this is a major difference between my setup and the one from @agramfort). I'll try Anaconda right now. Be right back.

Eric89GXL, 2017-05-23:
> I'll try Anaconda right now. Be right back.

If you have problems let us know. Our installation instructions promote Anaconda, so if it's painful, we need to make it easier.

cbrnr, 2017-05-23:
OK, I tried Anaconda, and it does work. However, it really seems to be related to MKL, or rather, not having MKL and therefore OpenBLAS. If I `conda install nomkl`, I get the same segfault.

Eric89GXL, 2017-05-23:
So even worse (harder to fix) than a `sklearn` bug, it's probably an OpenBLAS bug :(

cbrnr, 2017-05-23:
So what do you suggest? Open an issue for scikit-learn and see what they say? Report the issue for OpenBLAS or LAPACK (no idea which of the two is causing the issue)? I'll try to replicate the issue on my Linux machine tomorrow. In any case, this is not good because other than speed there should be no differences between using MKL or not.

Eric89GXL, 2017-05-23:
>  Open an issue for scikit-learn and see what they say?

Might as well so they are aware. They might have some ideas, or maybe they did something abnormal that they can fix.

> Report the issue for OpenBLAS or LAPACK.

IIUC OpenBlas provides an implementation of LAPACK functions, so it should be OpenBLAS's problem, assuming that the sklearn people (or SciPy if it wraps there) aren't doing something ugly in their call. So I'd wait to hear what the sklearn people say before going to the OpenBLAS folks. You'll probably need to keep minimal-izing the example at each stage, so `sklearn` is also an easier stop to make on your way to OpenBLAS.

> In any case, this is not good because other than speed there should be no differences between using MKL or not.

True, but [even MKL has had its own unresolved crashing bugs (for years!)](https://software.intel.com/en-us/comment/1904961), so these are just standard perils of relying on external software at all.

(Also keep in mind that there are important differences between OpenBLAS and MKL in terms of licensing, but I don't think that's what you mean.)

cbrnr, 2017-05-23:
Thanks @Eric89GXL, this sounds like a good plan 👍 ! I'll link this issue here in case you're interested.

I agree that such problems are standard and I guess it's our job to report them and try to get them fixed so that as many people (configurations) as possible can use our code. And yes, I wasn't referring to licensing, but I don't like MKL in general for not being open source.

In the meantime, would it be sensible to change the PCA solver in MNE's ICA fit method to `full`?

Eric89GXL, 2017-05-23:
I don't know the code that well, but my guess is that the default solver
was chosen for efficiency. Hopefully your crash is the exception, not the
rule (like my linked MKL problems have been).

What do you think about instead making a documentation update to that
parameter or the notes section stating something like "on some
configurations (e.g., state yours succinctly) the default fails, it is safe
to use full." Would that have helped you?

cbrnr, 2017-05-24:
Seems like it only occurs on macOS in a non-MKL Python env. It works on Anaconda for Windows (where there is no `nomkl` option) and on Linux (using system Python and Anaconda with both MKL and `nomkl`).

Putting this in the docs is a good idea, but let's wait until we hear back from farther upstream. If this really cannot be solved quickly, I'd prefer exposing the `svd_solver` argument so that users can more easily change it if they run into this issue.

agramfort, 2017-05-24:
>
> default solve was chosen for efficiency.
>

yes

cbrnr, 2017-05-24:
sklearn issue: https://github.com/scikit-learn/scikit-learn/issues/7626
which is really a
scipy issue: https://github.com/scipy/scipy/issues/7131
which is really an issue related to Apple's Accelerate framework. 😞  so there seems to be little hope for a fix anytime soon. This is too bad, because many people won't be using Anaconda on a Mac...

agramfort, 2017-05-24:
then it's not an openblas problem. You should not use Accelerate ! it's
slow and has issues as you've seen... use openblas or MKL for serious
numerics.

cbrnr, 2017-05-24:
Yes, if only it were that easy on a Mac 😄  - right now (meaning before scipy 1.0), avoiding Accelerate means using Anaconda MKL as the _only_ option. @Eric89GXL I think a note in the docs would be very helpful! Where would you mention this? Since this happens when a user tries to perform ICA, it should probably mentioned there.

agramfort, 2017-05-24:
I've had the problem before on my tasks not just ICA.

that's why we recommend Anaconda. For me install is really painless

cbrnr, 2017-05-24:
Installation of Anaconda is not the problem, there are just some less ideal things such as installing everything completely separate from the system (which of course is also an advantage) - when you `conda install rpy2` you automatically get a separate R environment pulled in into Anaconda (I wanted to use my existing R installation). Many libraries such as Qt get installed, and these can get in the way of manually installed libs, especially when you're developing with C++/Qt and you're not aware that Anaconda ships its own versions, etc. etc. Anyway, the upside is that installation is really easy and most things just work very well. I guess I'll have to switch back at least until this issue is resolved.

Eric89GXL, 2017-05-24:
> there are just some less ideal things such as installing everything completely separate from the system (which of course is also an advantage)

If you want, you can use `conda` (not Anaconda) to get some minimal system going with numpy/scipy/sklearn, and install everything else yourself with pip like you have otherwise been doing. Then I think the only sandboxed libs would be mkl.

cbrnr, 2017-05-24:
You're right, I was always using Miniconda instead of Anaconda, but I've always installed packages via `pip` only if they were not available in `conda.` Good idea, these are more or less the only sandboxed libs then (except for the libs Python need  such as openssl, tk, and zlib, but that's OK).

cbrnr, 2017-05-24:
Sadly, this is not that easy - seems like if you use conda you'd better use Anaconda packages. For example, matplotlib installed via pip is not working because Anaconda Python was not built as a framework. Same problem with other packages such as wxpython Phoenix (which is only available via pip). Now I remember why I decided to switch to Homebrew Python...

Eric89GXL, 2017-05-24:
Ahh bummer. OSX makes things complicated :(

agramfort, 2017-05-24:
I have anaconda and for me it's simple :)

cbrnr, 2017-05-30:
On a related note, isn't it sufficient to use MKL-versions of numpy and scipy? Or do I need a specific scikit-learn-mkl package? My thinking is that if I have MKL-enabled numpy and scipy packages, other packages using numpy/scipy should automatically benefit from the speedup. Or are there any functions using LAPACK/BLAS that are specific to scikit-learn?

agramfort, 2017-05-30:
use MKL-versions of numpy and scipy should be enough

cbrnr, 2017-07-20:
Closing since this is caused by Apples's Accelerate framework. I'm hoping they'll eventually fix it, but until then Mac users are forced to use MKL or OpenBLAS (and the easiest way to do that is to use Anaconda).

closed 2017-07-20 by cbrnr