mne-tools/mne-python #4283: Bug read_raw_brainvision

agramfort, 2017-05-25:
from mailing list

```
I am using mne.io.read_raw_brainvision to read bipolar EMG and EKG data saved as brainvision files (vhdr).
I have 4 channels: EMGleft,EMGright,EKG and STI 014 with the triggers.

However, when I import the data in MNE python, the first three channels are not correctly imported.
They all look like a combination of EMG data, however they do not correspond to the original data, particularly even the third channel is not at all EKG. 
You can see attached to the email the images with a portion of imported data from the three channels.

Is it possible that the function is using a reference by default? I don't see this option in the function.
This is how I am using the function:

emg = mne.io.read_raw_brainvision(emgFname,montage=None, misc='auto',scale=1.0,preload=True, verbose=True)
```

answer by @bburle 

```
Investigating the issue, it appears that mne.io.read_raw_brainvision is
currently not using the information about data orientation in the header
(.vhdr) file:

DataOrientation=VECTORIZED

Apparently, the function assumes that data are multiplexed. It's just a
matter of transposing the matrix while reading. I'll try to pinpoint the
lines in mne.io.read_raw_brainvision where this should be done.
```

problematic file is here

[EMG.zip](https://github.com/mne-tools/mne-python/files/1028311/EMG.zip)

bburle, 2017-05-26:
Being not familiar with the "_BaseRaw" object, I haven't (yet) identified where the problem is. But to advance a bit, here is a short piece of code that 1) replicate the problem, 2) open the file correctly (through direct memmap). I haven't been able to get the same (wrong) data as mne through a direct memmap, by transposing and/or reshaping differently the array. Hope this can be useful...

```
import numpy as np
import pylab as plt
import mne

emgFname = 'sb1b1'

mne_data = mne.io.read_raw_brainvision(emgFname+'.vhdr',montage=None, misc='auto',scale=1.0,preload=True, verbose=True)
plt.subplot(311)
plt.plot(mne_data._data[0,22000:120000]) #portion of data wheres something interesting occurs
plt.subplot(312)
plt.plot(mne_data._data[1,22000:120000])
plt.subplot(313)
plt.plot(mne_data._data[1,22000:120000])
#wrong data

raw_data = np.memmap(emgFname+'.eeg', np.float32)
rs_data = raw_data.reshape((3,813056)) #hard coded based on the vhdr file
plt.subplot(311)
plt.plot(rs_data[0,22000:120000])
plt.subplot(312)
plt.plot(rs_data[1,22000:120000])
plt.subplot(313)
plt.plot(rs_data[2,22000:120000])
# data OK

```

jona-sassenhagen, 2017-05-30:
Is this on master? IIRC we had a problem with long channel names earlier that should now be fixed. WIll try to look at it later, but might take a while :(

@teonbrooks

palday, 2017-05-31:
There's a small copy-paste error in the example:

```python
mne_data = mne.io.read_raw_brainvision(emgFname+'.vhdr',montage=None, misc='auto',scale=1.0,preload=True, verbose=True)
plt.subplot(311)
plt.plot(mne_data._data[0,22000:120000]) #portion of data where something interesting occurs
plt.subplot(312)
plt.plot(mne_data._data[1,22000:120000])
plt.subplot(313)
# this index needs to be a 2
plt.plot(mne_data._data[2,22000:120000])

# reshaping by hand
raw_data = np.memmap(emgFname+'.eeg', np.float32)
rs_data = raw_data.reshape((3,813056)) #hard coded based on the vhdr file
plt.subplot(311)
plt.plot(rs_data[0,22000:120000])
plt.subplot(312)
plt.plot(rs_data[1,22000:120000])
plt.subplot(313)
plt.plot(rs_data[2,22000:120000])
# data OK
```

palday, 2017-05-31:
The fix is relatively straightforward for this example; the trick is that the function that has to be modified isn't in the `RawBrainVision` class or even `brainvision.py`.

PR #4293 (2017-06-29): MRG+1: fix data orientation issue in BV data
This is the initial work on fixing issues related to data-orientation in the BV-reader. It fixes the bug based on the (corrected) example given in: #4283.

To do:

- [ ] ~~write tests for changes to `_read_segments_file()` (incl. add dataset to test repository)~~
- [ ] ~~handle this issue when data is read using `_read_segment_file()`~~
- [ ] ~~write tests for changes to `_read_segment_file()` (incl. add dataset to test repository)~~
- [x] write test for reading vectorized data
 
Closes #4283.
closed 2017-06-29 by jaeilepp