mne-tools/mne-python #4284: Example images to show in the front-page slider

choldgraf, 2017-05-25:
Now that we've got a beautiful front-page slider on the website:

http://martinos.org/mne/dev/index.html

Are we happy with the images that are there? I think we should cover all of the major ephys analysis steps in there as images (at least the ones that produce pretty plots). One that I think is missing is a [time-frequency decomposition](http://martinos.org/mne/dev/auto_examples/time_frequency/plot_source_label_time_frequency.html#sphx-glr-auto-examples-time-frequency-plot-source-label-time-frequency-py) of the signal. Also maybe something like the [cortical parcellation](http://martinos.org/mne/dev/auto_examples/visualization/plot_parcellation.html#sphx-glr-auto-examples-visualization-plot-parcellation-py) example since it's pretty?

Any other things we're missing?

choldgraf, 2017-05-25:
pinging @dengemann @kingjr @Eric89GXL @agramfort in case they have thoughts (I'm happy to open the PR myself, just asking for feedback :-) )

mmagnuski, 2017-05-26:
Most likely a side topic: when did some of the example images stopped being pleasantly pixelated? I am referring to images for Machine Learning and Encoding Models (which do not show up when you follow the link BTW...).

choldgraf, 2017-05-27:
@Eric89GXL made that change in a recent PR. It's using linear interpolation to make things smoother now.

Eric89GXL, 2017-05-27:
The cortical parcellation is really more of a PySurfer thing but why not. TF is definitely missing, if we have a good example let's add it. Otherwise nothing else from me.

@mmagnuski the reason I changed it is that linear interp in 2D is the equivalent of using `plt.plot` instead of `plt.step` for 1D plots. If you expect your signals to be smooth it at least makes some sense.

agramfort, 2017-05-27:
I updated the web site with a new rsync

choldgraf, 2017-05-27:
looks to be working on my end now!

agramfort, 2017-05-27:
great !

kingjr, 2017-05-30:
I would add a non-folded brain, a helmet/head plotting and a raw signal.

And I would have a separate folder for these images such that we can make ones that are of the right shape colors etc, rather than suboptimal illustrations from the example .

Eric89GXL, 2017-05-30:
> And I would have a separate folder for these images such that we can make ones that are of the right shape colors etc, rather than suboptimal illustrations from the example .

This creates more maintenance burden than the current solution (which is a sort of "minimum required maintenance" carousel), so if we go this route I think we'd need to have someone who volunteered for this job and fixed it when things broke

larsoner, 2019-08-20:
Closing, we are happy enough :)

closed 2019-08-20 by larsoner