mne-tools/mne-python #4288: use nilearn.plotting.plot_surf for plotting surface data

jona-sassenhagen, 2017-05-31:
Nilearn can plot static surface data in matplotlib (and thereby, the notebook and py3). Considering how often Mayavi drives @kingjr into an existential crisis, maybe adding this as an alternative is worth a try.
The code isn't complicated.

http://nilearn.github.io/modules/generated/nilearn.plotting.plot_surf.html

agramfort, 2017-05-31:
see https://github.com/mne-tools/mne-python/pull/4086

kingjr, 2017-05-31:
I also like the 3-view glass brain of nilearn for a diagnostic (non
interactive) plotting

agramfort, 2017-05-31:
sounds like we have a volunteer :)

jona-sassenhagen, 2017-05-31:
I have to finish a bunch of other PRs first sadly.

jona-sassenhagen, 2017-06-01:
See https://github.com/nilearn/nilearn/issues/1461

Minimally, we could just give an example and possibly a bit of glue for integration with nilearn.

jona-sassenhagen, 2017-06-01:
(Essentially, mapping from an stc or labels to the stat_map of nilearn's plot_surf would be super low effort and would allow quick visualisation of surface locs.

I agree with @kingjr  on the glass brain.)

agramfort, 2017-07-21:
there is now an stc.plot with matplotlib backend

closed 2017-07-21 by agramfort