mne-tools/mne-python #4290: What does raw.info["events"] contain?

cbrnr, 2017-05-31:
When I load a BDF file, I need to call `mne.find_events(raw)` in order to extract the events found in the stim channel. However, I found that there is a key called `events` in the `raw.info` dictionary (which is usually empty for my files). What exactly is supposed to be in this list of dicts (according to the [docs](https://martinos.org/mne/stable/generated/mne.Info.html#mne.Info))? Is this field somehow related to the events I find with `mne.find_events`?

Eric89GXL, 2017-05-31:
Good question, I've never fully understood it either. Apparently it can be a list of dict, where each dict has entries "channels" and "list", such as:
```
>>> raw = mne.io.read_raw_fif(mne.datasets.testing.data_path() + '/SSS/TRIUX/triux_bmlhus_erm_raw.fif')
>>> print(raw.info['events'])
[{'channels': array([324], dtype=int32), 'list': array([     7,    256,  16640, ..., 329992,    256,  16640], dtype=int32)}]
>>> raw.ch_names[324]
u'SYS201'
>>> mne.find_events(raw, 'SYS201')
array([], shape=(0, 3), dtype=int32)
>>> raw[324, :][0][0]
array([ 0.,  0.,  0., ...,  0.,  0.,  0.])
```
Or assuming 1-based indexing:
```
>>> raw.ch_names[323]
u'STI101'
>>> raw[323, :][0][0]
array([ 256.,  256.,  256., ...,  256.,  256.,  256.])
>>> mne.find_events(raw, 'STI101')
array([], shape=(0, 3), dtype=int32)
>>> mne.find_events(raw, 'STI101', mask=256)
array([], shape=(0, 3), dtype=int32)
>>> raw[323][0][0].max()
256.0
>>> raw[323][0][0].min()
256.0
```
So at the very least our description seems inaccurate.

The description in the official FIFF definitions is:

> "Information about the events detected on the trigger channels during data acquisition."

And for the subtype `list`:

> "List of events, 3 integers per event: [number of samples, before, after]"

We should probably change it to contain this, plus say that it's typically not used in the MNE workflow and `mne.find_events` is preferred.

When someone gets bored they could actually go through and beef up that section by using the FIFF constant definitions. I can try at some point when I need a science break.

agramfort, 2017-05-31:
I have no idea to be honest

@mshamalainen any idea?

cbrnr, 2017-06-01:
Thanks @Eric89GXL - mentioning that `mne.find_events` is preferred would be very helpful! It seems like several reading functions automatically parse events from a stim channel, and this information ends up in `raw.info["events"]`. Would it make sense to remove this field completely to avoid redundancy (and potential confusion)? What mechanisms are currently using this field? Could we fill it with the result of `mne.find_events` instead?

agramfort, 2017-06-01:
I'd like to know why ["events"] was there in the first place...

personally i've never used it.

jjnurminen, 2017-06-01:
I've heard that the the Elekta acquisition system parses trigger channels and writes at least some event info into the fiff, in addition to the raw trigger data. I never tried to use it though. Maybe Matti can confirm (or deny).

jona-sassenhagen, 2017-06-01:
Can we use this to store some events which we can't store in events channels, eg those with overlapping time points?

jjnurminen, 2017-06-01:
If you write the extra events back into fiff, some other software may get confused by them, at least in theory. However I'm not sure if even Elekta software uses the events info for anything.

mmagnuski, 2017-06-02:
I also got confused by `raw.info['events']` some time ago: #3380 and stored my events array there, at that time @agramfort said it would be lost on write (although we didn't test it then).

agramfort, 2017-06-03:
indeed but it's not the same data structure...

PR #4297 (2017-06-03): MRG: Update Info docstring
Simple stuff:
* Updates `Info` docstring to be much more complete.
* Fixes some minor documentation bugs.

For discussion:
* Reorganizes some classes in the doc. The way we had it before, we had a "Classes" heading that only had some of the classes. This has been removed in favor of "Most-used classes", which has just Raw, Epochs, Evoked, and Info, all other classes have been moved to their respective sections. The alternative I see is to move *all* classes to the top Classes section. It seems a bit strange to have classes (partially or totally) centralized in one pl [... cut]
closed 2017-06-03 by agramfort