mne-tools/mne-python #4302: Are projections applied when fitting ICA?

cbrnr, 2017-06-03:
I'm computing ICA on continous raw EEG data. In my pipeline, I set

    eeg.set_eeg_reference()

before doing

    ica = ICA(method="extended-infomax", random_state=1)
    ica.fit(raw)

Does this imply that I fit ICA on average referenced `raw` data? Or do I need to explicitly invoke `raw.apply_proj()` before fitting ICA?

Similarly, if I only have the three lines described above, when I call `raw.get_data()`, is the average reference projection applied automatically?

From what I've seen in the code, it seems that the average reference projection is _NOT_ applied automatically except for in `raw.plot` (where it can be switched on and off).

agramfort, 2017-06-03:
> I'm computing ICA on continous raw EEG data. In my pipeline, I set
>
> eeg.set_eeg_reference()
>
> before doing
>
> ica = ICA(method="extended-infomax", random_state=1)
> ica.fit(raw)
>
> Does this imply that I fit ICA on average referenced raw data? Or do I
need to explicitly invoke raw.apply_proj() before fitting ICA?

you need to call apply_proj AFAIK

> Similarly, if I only have the three lines described above, when I call
raw.get_data(), is the average reference projection applied automatically?

look at raw.info['projs']
it will tell you if proj is active (ie applied) or not

> From what I've seen in the code, it seems that the average reference
projection is NOT applied automatically except for in raw.plot (where it
can be switched on and off).

yes

cbrnr, 2017-06-06:
Thanks, that's good to know. There's a note in the documentation of `set_eeg_reference` that average reference is actually the default if nothing else is specified, and that `apply_proj` needs to be called on preloaded data. However, this makes me wonder which functions actually use the average reference projector with preloaded data. None (i.e. do I always have to apply the projector)?  Would ICA use the average reference projection if I didn't preload the data?

agramfort, 2017-06-06:
I have no strong feeling here. Please make suggestions of improvements
(code and/or documentation)

cbrnr, 2017-06-06:
Ideally, the docs should state when the average reference projector is automatically applied. Since I don't know if this is the case I can't suggest specific changes. If it is the case that the projector is _never_ applied for preloaded data, the [last point in the notes section](https://martinos.org/mne/stable/generated/mne.set_eeg_reference.html) could be reformulated as "Be aware that on preloaded data, SSP projectors are never automatically applied. Use the `apply_proj()` method to apply them." (replace "not" with "never").

cbrnr, 2017-06-06:
Hm, just found https://github.com/mne-tools/mne-python/issues/3998 and https://github.com/mne-tools/mne-python/issues/4004. It might just be me, but re-referencing to average reference is still not very intuitive. @hoechenberger do you think your concerns have been addressed?

hoechenberger, 2017-06-07:
Hi @cbrnr, #4004 has certainly improved things, personally I still would not call it intuitive. Average-referencing behaves different than other referencing schemes. I believe that is the main issue, and _all_ of my colleagues who have tried out MNE had trouble with this. We all come from an EEG-only background (ok, some of us have done fMRI too; but there's no MEG people here), and without exception we all found this confusing. #4004 has greatly clarified the docs, but the term "SSP projector" alone is unknown to all EEG folks I've asked. I don't know how to solve this issue really.

cc @lorrandal

cbrnr, 2017-06-07:
Thanks @hoechenberger, same here! The single most confusing behavior for me is that `raw.set_eeg_reference()` does not actually re-reference the data to average reference. What's more, this might only be true if the data is preloaded. I actually wanted to compute ICA on average referenced data, and later noticed that the data was never re-referenced.

Since there still seems to be room for improvement, how could we make re-referencing more intuitive especially for non-MEG folks? The simplest solution seems to be making average reference a true re-referencing operation instead of a projector. What are the arguments in favor of keeping it a projector?

hoechenberger, 2017-06-07:
x-ref to related mailing list posting: https://mail.nmr.mgh.harvard.edu/pipermail//mne_analysis/2017-June/004094.html

cbrnr, 2017-06-07:
It's weird that bad channels were causing issues with ICA, because `ICA.fit` ignores/excludes all bad channels, independent of the reference. I'll respond to this message on the list once I'm subscribed 😄 .

Eric89GXL, 2017-06-07:
> without exception we all found this bit confusing.

Okay so we clearly still have a problem that we need to fix.

One thing we could consider is adding `set_eeg_reference(..., apply=True)`:

1. Deprecation cycle in 0.15 where default is `apply=None`. This means True for non-avg and False for avg, with a warning either way. MEG people will probably want to do `apply=False` and EEG people will probably want to do `apply=True` for a cycle to avoid the warning.
2. In 0.16 the default is `apply=True`. I *think* everything will still work okay, even for MEG people if they don't remember to do `apply=False`.

@agramfort WDYT?

cbrnr, 2017-06-07:
If you apply a reference, including average reference, is it possible to undo this (i.e. re-reference to a different ref)?

Eric89GXL, 2017-06-07:
Yes you should be able to re-reference. I *think* our API allows it even if you've applied an EEG reference as a projector, but if it doesn't, we can probably make it support it.

Also cc @wmvanvliet about the above proposal for `apply=True`

hoechenberger, 2017-06-07:
> If you apply a reference, including average reference, is it possible to undo this (i.e. re-reference to a different ref)?

Yes, this certainly works. Or at least it used to -- haven't tested with the latest master :)

hoechenberger, 2017-06-07:
@Eric89GXL 
I like your proposal with `apply=True`, seems much more sensible to me!

wmvanvliet, 2017-06-07:
What would `set_eeg_ref(raw, ['EEG1', 'EEG2'], apply=False)` actually *do*?

mmagnuski, 2017-06-07:
> Yes you should be able to re-reference. I think our API allows it even if you've applied an EEG reference as a projector, but if it doesn't, we can probably make it support it.

@Eric89GXL - if I'm not mistaken I had a situation recently when I first re-referenced to avg and applied proj, then marked bad channels and interpolated them and wanted to re-ref to avg again and mne told me I cannot because the avg ref is already there... But I could have preformed epoching in the meantime which has `proj=True` by default (which BTW adds to the confusion). I can check it later today.

Eric89GXL, 2017-06-07:
> What would set_eeg_ref(raw, ['EEG1', 'EEG2'], apply=False) actually do?

This would be an error. `apply=False` would only be allowed in avg mode.

> I could have preformed epoching in the meantime which has proj=True by default (which BTW adds to the confusion).

Yes this is related to #2310, which I started addressing a long time ago but it's a pain

cbrnr, 2017-06-07:
But this still doesn't get rid of treating references based on a list of channels and the average reference differently. After all, we could express the average reference with a list of all channels. I'd vote for treating all references the same under the hood (but shorthand notations can of course remain, such as `None` for average reference - although this is another rather unintuitive choice and should probably be changed to `[]`).

Eric89GXL, 2017-06-07:
> But this still doesn't get rid of treating references based on a list of channels and the average reference differently.

To the user, or under the hood? Hopefully to the user it's pretty transparent / they won't notice. Under the hood it needs to be different, because the inverse modeling code handles (and requires actually) an average reference projection for EEG data.

In other words, I agree it's not as simple as what you propose, but what you propose unfortunately breaks our inverse code.

cbrnr, 2017-06-07:
OK, if existing code relies on the method/function `set_eeg_reference` to behave exactly like this, we could introduce a new argument value `"average"` to mean an average reference which gets applied by default. Just like all other references consisting of a list of channels.

If the behavior can be changed, would a separate function that computes/sets an average reference _projection_ be an option?

Eric89GXL, 2017-06-07:
> If the behavior can be changed, would a separate function that computes/sets an average reference projection be an option?

I'd rather just add a kwarg to `set_eeg_reference`. I guess this would mean an alternative proposal like `set_eeg_reference(..., projection=False)` where inverse imaging folks would want to use `projection=True` (which would also imply it wouldn't be applied immediately), and `projection=True` would only be allowed with `None` or whatever we change it to if we do change it (e.g., `'average'`). I'm okay with that, too, I suppose.

cbrnr, 2017-06-07:
A `projection` kwarg makes more sense to me - good idea! This way, all references would behave the same to the user (i.e. they get applied by default). Under the hood, `'average'` (or whatever it will be called) in combination with `project=False` could be implemented by creating a reference with a list of all channels (excluding bads). When `project=True`, the current implementation would be used. Just let's make sure that all re-referencing operations are reversible (i.e. I want to be able to switch to different references in my pipeline based on the same `raw` object), but as many of you have already stated this is already possible.

Eric89GXL, 2017-06-07:
> Let's just make sure that all re-referencing operations are reversible

Yes agreed. Although technically this is (hopefully?) an orthogonal issue to the proposed API changes anyway, so I'm happy to punt on that for now :)

So the proposal for `set_eeg_reference` seems to have converged to:

1. Add `projection=False` default (via deprecation+warning, 0.15 None->True). By 0.16 this makes things easier for EEG people, but harder for source localization / inverse imaging people. It unifies how `set_eeg_reference` works. `projection=True` only works in avg mode.

2. Change `ref_channels=None` to `ref_channels='average'` to be explicit.

@wmvanvliet @hoechenberger @mmagnuski @agramfort WDYT?

hoechenberger, 2017-06-07:
@Eric89GXL This actually sounds very good. The API would become more explicit _and_ more intuitive at the same time. I really like that ;)

@lorrandal What do you think?

agramfort, 2017-06-07:
can you write down explicitly the new API code for

- set avg ref but don't apply a proj (just add it to info['projs'])
- set avg ref and apply it (is it always done with a proj?)
- set custom reference
- undo any of the referencing above

thx

Eric89GXL, 2017-06-07:
>  set avg ref but don't apply a proj (just add it to info['projs'])

`set_eeg_reference(inst, projection=True)` (docs would specify `projection=True` is special because it does not immediately apply it, and uses projection math during projection application stages instead)

> set avg ref and apply it (is it always done with a proj?)

* For non-projection method: `set_eeg_reference(inst)`
* For projection method: `set_eeg_reference(inst, projection=true).apply_proj()`

> set custom reference

`set_eeg_reference(inst, ref_list)`

> undo any of the referencing above

This is unchanged, same as in `master`. You can call `set_eeg_reference` with whatever arguments you want. We don't currently provide a way to un-reference EEG-referenced data, even though in the case of an unapplied projection vector it is possible (well get it back to the nose ref at least), using this function. But I guess you can do `inst.del_proj(...)` with the EEG projection vector selected.

agramfort, 2017-06-07:
ok

what does set_eeg_reference(inst) does? can I still use inst for inverse
modeling?

mmagnuski, 2017-06-07:
@Eric89GXL I like what you're proposing, but wanted to ask a related question related to avg proj and inverse modeling (@agramfort too):
I understand that it's better to have avg ref for inverse modelling but why does mne require avg ref to be a projection? And a related question: if I have some data with average ref already applied would mne's inverse accept if I added a fake avg ref proj (fake because the average would be already zero) or would I have to re-ref to some other channel and add avg ref proj?

Eric89GXL, 2017-06-08:
> what does set_eeg_reference(inst) does? can I still use inst for inverse modeling? 

Nope, you need `set_eeg_reference(inst, projection=True)`.

This is the unfortunate tradeoff currently for making things consistent with all other `set_eeg_reference` use cases and making the package more EEG- (and ECoG- and ...) friendly. FWIW to the extent that it makes it less M/EEG / NeuroMag-friendly, it is also making us M/EEG people be more explicit, which I don't mind. The frequency with which this issue of projections comes up is more troubling to me.

> why does mne require avg ref to be a projection

From the practical side, all of the inverse code accounts for things using projections (and compensation matrices). Actually come to think of it we could hack together EEG references as a form of compensation matrices I think -- but this would be even uglier than what we have now.

From the theoretical side, I seem to remember someone saying avg ref leads to less error but I can't remember offhand. @agramfort?

If it is possible to make the inverse modeling code handle non-average projections, this would also take care of @agramfort's point about `set_eeg_reference(inst)` not working, as we could make the code handle this case properly... but it would be a lot of work in practice (we would need to make very sure we did it correctly).

> if I have some data with average ref already applied would mne's inverse accept if I added a fake avg ref proj (fake because the average would be already zero)

Well this wouldn't really be "fake" per se -- it would be a real projection operator you'd be adding. Projection operations can be repeated over and over with no effect (beyond small amplification of numerical noise). So although it wouldn't really change the data, it would provide proper accounting for what actually happened to the data during inversion.

cbrnr, 2017-06-08:
Coming back to the question of undoing referencing, EEG data are always measured with respect to _some_ reference. However, if we load a data set this information might not be present in the file, so we should have a way to specify the reference manually. For example, BioSemi records all channels with respect to the CMS electrode (and it uses an additional DRL electrode to improve the CMRR). Other systems might record channels with a different (more conventional) reference such as Fz. Therefore, we should be able to let MNE know which reference is associated with the data. By default, this value could be `"unknown"` or something. I think currently `set_eeg_reference([])` means "don't change anything", which is what we want, but we might want to add some value to specify what this current reference is.

hoechenberger, 2017-06-08:
@cbrnr 
I just took a look at our latest experimental analysis code -- "experimental" meaning I didn't have time to thoroughly check if everything is _actually_ working as intended. But basically, our situation is the following:

We use an ActiChamp system, usually with Oz set as reference during recording. When we import the data into MNE, the Oz electrode is missing. We would filter and epoch the data, run ICA, and only then try to recover the channel and reference to average:
```python
# Recover Oz electrode, which was used as a reference during recording.
mne.add_reference_channels(epochs, 'Oz', copy=False)
epochs.set_montage(montage=montage)
epochs.info['custom_ref_applied'] = False

# Average reference
epochs.set_eeg_reference(ref_channels=None)
epochs.apply_proj()
```

cbrnr, 2017-06-08:
Thanks @hoechenberger - adding a reference channel means adding a zero channel called `Oz` in your case, right? I'm not sure if you want to add a zero channel before calculating the average reference though...

hoechenberger, 2017-06-08:
> Thanks @hoechenberger - adding a reference channel means adding a zero channel called Oz in your case, right? 

Yes.

> I'm not sure if you want to add a zero channel before calculating the average reference though...

The reference is always "zero" by definition; applying the avg ref essentially "recovers" that channel's information, if you will. At least that's my understanding :)

cbrnr, 2017-06-08:
Hm, is this procedure compatible with what is described [here](https://sccn.ucsd.edu/wiki/Chapter_04:_Preprocessing_Tools#Re-referencing_the_data)?

hoechenberger, 2017-06-08:
> Hm, is this procedure compatible with what is described here?

I think it's essentially identical to what is described in the section starting with `Re-referencing data can be more complicated.`

cbrnr, 2017-06-08:
😄  yes, this is really true! So essentially you have to decide if you want to put the current reference electrode back in the data or not - in your case, you want Oz to be part of the data, and in my case (Biosemi), I don't want the current reference(s) CMS (and DRL) in the data, so I'd not add any reference channels. Correct?

What does this mean for the proposed API change? Are we still good with introducing a `projection` argument in `set_eeg_reference`?

hoechenberger, 2017-06-08:
> yes, this is really true!

... but also really hard to understand at first, because the EEGLAB docs are ... errm... kind of confusing in this regard.
Will respond to your other points later today.

agramfort, 2017-06-08:
from a user perspective, if I now want to do inverse modeling and I did
set_eeg_reference(inst, projection=False) I will get an error telling me to
set projection=True? or can we automatically add the eeg avg ref in inverse
code if custom_ref is False? that would make it easier.

and yes avg ref is the standard way for inverse modeling. The reason it's
done with projection is that you need to apply the proj to forward solution
and noise cov in the inverse code.

cbrnr, 2017-06-08:
Agreed. Ideally this should be handled transparently in the inverse code if it always needs to project to average ref anyway. No need to "pollute" the user-facing method to set a reference IMO (that's also how it is done in EEGLAB with DIPFIT BTW).

Where is the information on the currently used (applied) reference (including average reference) stored?

agramfort, 2017-06-08:
avg proj is on if you have in info['projs'] a Projection instance called
"EEG average reference" marked as "active"

cbrnr, 2017-06-08:
And other refs? Are these stored in `raw.info['custom_ref_applied']`? Should there be one field containing the currently set reference instead? I think it would be nice if this was not spread across at least two different fields.

wmvanvliet, 2017-06-08:
We don't have to break any existing code. When doing `set_eeg_reference(inst)` or `set_eeg_reference(inst, projection=False)`, we need to do the right thing and keep `info.custom_ref_applied=False`.

The "custom ref" represents something *other* then an average ref. When doing inverse computations, `_needs_eeg_average_ref_proj` first checks the `custom_ref_applied` flag and then proceeds to check for the presence of an average ref proj. Since one is not present, an average ref projection will be added, which is exactly what we want.

lorrandal, 2017-06-08:
@cbrnr @hoechenberger 
> in your case, you want Oz to be part of the data

This was exactly our situation, since we wanted to compare this dataset with another that was acquired keeping all 64 electrodes -- Oz included.

> identical to what is described in the section starting with Re-referencing data can be more complicated.

Should be this, in my understanding information regarding Oz is not lost but simply spreaded across all other electrodes. So adding Oz with zero values and re-referencing would make us able to ricover that data.

mmagnuski, 2017-06-08:
@Eric89GXL @agramfort 
Thanks for clarifying!

agramfort, 2017-06-08:
seems like a plan? @cbrnr do you give it a try?

cbrnr, 2017-06-08:
I will need to wrap my head around the existing API first, and I can't do it immediately because I'm a bit swamped with other stuff, but yes, I will try to push a PR to address this (if there are not other volunteers who want to do it right now :-)).

Eric89GXL, 2017-06-08:
> Coming back to the question of undoing referencing, EEG data are always measured with respect to some reference. However, if we load a data set this information might not be present in the file, so we should have a way to specify the reference manually

FYI this currently lives in `info['chs'][n]['loc'][3:6]`. For Neuromag EEG it's on the nose.

> We don't have to break any existing code.

It's good that makes the `projection=False` change will make things easier for MEG people. The disadvantage is that we've lost any record-keeping that the data are no longer referenced to the nose. Perhaps we need to do something with `info['chs'][n]['loc'][3:6]` to fix this. Even setting it to zero might be okay. I'd need to check whether it messes with the forward code -- I don't think it should, because the ref is subtracted during forward computation, but then effectively added back in (cancelling this out) during avg ref.

> yes, I will try to push a PR to address this

That would be great. I might have time to try at some point, too. We'll see who has time first :)

Eric89GXL, 2017-06-26:
Related: #4352.

cbrnr, 2017-07-05:
Quick reminder, is the currently applied reference stored somewhere? `raw.info["custom_ref_applied"]` contains only a boolean value but not the actual channels. For example, if I set `raw.set_eeg_reference(["Cz"])`, where is this information stored?

wmvanvliet, 2017-07-05:
> For example, if I set raw.set_eeg_reference(["Cz"]), where is this information stored?
Nowhere.

cbrnr, 2017-07-06:
So why not add a field to the info structure?

agramfort, 2017-07-06:
we'll have to see how to serialize this info when saving data as the writing of the info is quite rigid.

@wmvanvliet @jona-sassenhagen presently there is no way to recover what channel was used for referencing?

cbrnr, 2017-07-06:
Another thing came to my mind. Instead of changing `raw.set_eeg_reference` as discussed above, why don't we create a new method `raw.create_projection` (or something like that) to create an average reference projection? This way, `set_eeg_reference` would always apply any re-referencing and there's no need for a new `projection` argument. On the other hand, `create_projection` would always create a projection for average reference which can be used in the inverse modeling. What do you think?

agramfort, 2017-07-06:
create_projection is not explicit

there are different mne functions that output projections

how do you educate a user who wants to use average ref + then inverse
modelling?

when thinking about API, I step into the shoes of someone who needs to
make/write
a tutorial. That's when you write an example or the doc of your software
that you realise
that what you propose is not clear or too complicated.

cbrnr, 2017-07-06:
> there are different mne functions that output projections

Why don't we use an existing function to create an average projection then?

> how do you educate a user who wants to use average ref + then inverse
modelling?

It's done automatically in the modeling. How do you tell EEG users how to set average reference? Same problem here.

> when thinking about API, I step into the shoes of someone who needs to make/write
a tutorial. That's when you write an example or the doc of your software
that you realise that what you propose is not clear or too complicated.

Right now re-referencing is too complicated because projections are involved. If you think the proposed `projection` argument is less confusing then let's go for that instead. If the defaults are set accordingly (so `projection=False`), then everything is fine for EEG people, but you still need to educate inverse modeling people that they need to set `projection=True`.

Anyway, this was just an idea, I'm happy about any improvement to re-referencing.

agramfort, 2017-07-06:
I don't have time to structure the ideas here sorry.

We need someone to take the lead on this by:

- drafting the planned API (almost done AFAIK)
- write the code with proper deprecation if needed and catching potential
user errors with clear warnings
- update the documentation

wmvanvliet, 2017-07-06:
> Why don't we use an existing function to create an average projection then?

we do. It's called `mne.proj.eeg_average_ref_proj`

cbrnr, 2017-07-06:
OK, so why the need to overload the `set_eeg_reference` function with this projection then?

wmvanvliet, 2017-07-06:
because we wanted to write "use the `set_eeg_reference` function to set the desired EEG reference. So, remind me again what the big problem is with the function as it is now?

Is it that an average reference is not applied directly but instead is added as a projection? Having it as a projection actually makes a lot of sense, but we need to have better documentation for it.

Why does it make sense? Well, to do an average reference properly, it is vital that bad channels are properly marked, otherwise all your channels are corrupted. Whenever you use an average reference, it is very important to check the result. A good way to do this is to put the projections in "delayed" mode, so when plotting the evoked potential, you can turn the average reference projection on and off, see the effect of the reference, and verify it behaves properly.

An upcoming tutorial on referencing should do the above, instead of immediately applying the average reference projection without checking.

cbrnr, 2017-07-06:
> because we wanted to write "use the set_eeg_reference function to set the desired EEG reference". So, remind me again what the big problem is with the function as it is now?

> Is it that an average reference is not applied directly but instead is added as a projection?

Yes.

> Having it as a projection actually makes a lot of sense, but we need to have better documentation for it.

> Why does it make sense? [...]

True, but a projection is not necessary to do this. If you immediately apply the average reference, but you add the previous reference channels to the data before you apply it, then you can also undo the average referencing. Bad channels can also be excluded.

I think the main problem with the current behavior is that most EEG people have no idea what an SSP projection is, and - this is the real problem - that the average reference projection is _not_ automatically applied. When I say `set_eeg_reference` I expect that I actually set the reference. I agree that adding a `projection=False` argument to `set_eeg_reference` more or less solves this problem once the default is `False`.

[5 more replies]

PR #4382 (2017-07-20): [MRG] Add projection argument to set_eeg_reference
Fix #4302 and #3998.

Added new argument `projection=False` to `set_eeg_reference`.
closed 2017-07-20 by Eric89GXL